j7126/OctoPrint-Dashboard #88: Bug on RPI clones

mamaich, 2019-10-09:
File __init__.py. Old code:
`try: #Check to see if we are on an RPI or not
  import RPi.GPIO as gpio
  on_rpi = True
except (ImportError, RuntimeError):
  on_rpi = False

if on_rpi == True:
    import Adafruit_DHT
`
If the RPI clone does not have Adafruit_DHT module - the plugin would crash. 
The fix is simple - move import inside try block:
`try: #Check to see if we are on an RPI or not
  import RPi.GPIO as gpio
  import Adafruit_DHT
  on_rpi = True
except (ImportError, RuntimeError):
  on_rpi = False
`

StefanCohen, 2019-10-09:
Thanks,

I want to understand this. The Clone can import RPi.GPIO but not Adafruit_DHT? Is that correct?

StefanCohen, 2019-10-09:
I have decided to remove the dependency to the DHT module and rely on the Enclosure plugin instead for the next version so this will become a moot point. That’s why I will close this report.

closed 2019-10-09 by StefanCohen