j7126/OctoPrint-Dashboard #89: OctoPrint Dashboard : request to show DD:hh:mm instead of hh:mm

Bzguy1973, 2019-10-09:
I would like Dashboard to show the remaining and passed time in dd:hh:mm (instead of hh:mm), and taking this information into account for the predicted end time.

Like passed/ time & time left would show for instance 

- 2 Days 13 hrs 44 min printed
- 4 Days 3 hrs 23 min left
- estimated end time : 11/10/19 13:23

StefanCohen, 2019-10-09:
Thanks for the feedback!

I would actually prefer to go with ISO 8601 everywhere if possible but the "Estimated Total Print Time", "Estimated Time Left" and "Current Print Time" fields are actually strings coming from OctoPrint core, so there is not much I can do about them as they are not guaranteed to be available to me in a parseable format. They can all be sent in "fuzzy time format" depending on user settings. The only time field I have control over is the ETA. That's the reason I can't implement this FR

closed 2019-10-09 by StefanCohen

Bzguy1973, 2019-10-09:
no problem Stefan, it was just a suggestion :)

I was not aware of ISO 8601 ;)

Thanks anyway !

isaacrlevin, 2021-01-08:
You could let the user supply their own datetime format. As a spoiled American, I want everything in freedom units :O
https://docs.python.org/3/library/datetime.html#strftime-and-strptime-format-codes