j7126/OctoPrint-Dashboard #90: Missing DHT11 sensor data

joeynuggetz, 2019-10-10:
Did the latest version break humidity/sensor? Was working fine a few days ago. Not getting any readings now. Would be nice if each had their own icon too and maybe some docs on whether to use the physical GPIO pin or GPI number. I just guessed and got it to work.

StefanCohen, 2019-10-10:
I've had no other reports of that and my DHT11 works fine on two separate systems with 1.10.2.
I'm regardless working on removing the current implementation of sensor support. It will be replaced with a binding to the Enclosure plugin since it supports a better variety of sensors. I plan to push that within the next couple of days.

StefanCohen, 2019-10-10:
WIP:

[image]

closed 2019-10-12 by StefanCohen