j7126/OctoPrint-Dashboard #91: Please stop updating EVERYDAY!

dgdimick, 2019-10-11:
I know, most of the time users are complaining about issues not being addressed, however, can you please limit the number of upgrades that are pushed out to maybe twice a week? It seem like everything I turn around there's another update to Dashboard, and I'm needed to reboot the RPi.

Thanks

StefanCohen, 2019-10-11:
Thanks for the feedback. This is a project under active development. Releases will happen frequently for some time until the project reaches maturity or I get tired of it. You have a number of options:

1. Don't update for every relase. There is no need to update unless you want to.
2. Uninstall it until it is mature. 
1. Fork it and set your own release schedule.

closed 2019-10-12 by StefanCohen

dgdimick, 2019-10-13:
> Thanks for the feedback. This is a project under active development. Releases will happen frequently for some time until the project reaches maturity or I get tired of it. You have a number of options:
> 
> 1. Don't update for every relase. There is no need to update unless you want to.
> 2. Uninstall it until it is mature.
> 3. Fork it and set your own release schedule.

Stefan,

Thanks for the reply, and I guess I could call this "first world problems", I know really lame complaining that a bit of code is being actively developed and supported. Part of my complaint, is EVERY update has something good in it, it's not like you are just fixing bugs and things like that, you're putting REALLY COOL stuff in the PlugIn, like the Layer Duration, is really nice to see, I find that 99% of the time, I'm sitting on the Octoprint website in the Dashboard tab. So it's not like "gee he fixed some stupid thing" it's not like "Man what cool thing is in on the Dashboard now", thus the need to update everytime you push anything out.

StefanCohen, 2019-10-13:
No problem. I'm a big fan of the [release early, release often philosophy](https://en.wikipedia.org/wiki/Release_early,_release_often). Some users like it while it bothers other and that's just the way it is.

Stay put for [1.11.0](https://github.com/StefanCohen/OctoPrint-Dashboard/wiki#1110) that I plan to release today if everything goes well.