j7126/OctoPrint-Dashboard #93: Remove Context menus

StefanCohen, 2019-10-11:
**Is your feature request related to a problem? Please describe.**
The context menus are too obscure to be useful. User don't know that they are there and they are worthless on touch devices.

**Describe the solution you'd like**
Simply remove them.

j7126, 2019-10-13:
I think it would be better to leave them in but have a setting to enable them which is disabled by default. Some people might still use them.

StefanCohen, 2019-10-13:
I have considered that but opted to actually remove them while it is still early and not too many users are affected. I added them with the intention to implement controls for temperatures and maybe also motion control but I have changed my mind and will not go that way with the dashboard. My full reasoning is explained in #70.

StefanCohen, 2019-10-13:
Fixed in 1.11.0

closed 2019-10-13 by StefanCohen