j7126/OctoPrint-Dashboard #96: Simple question:  How do I get back out of Full Browser Mode in 1.1.1 release?

PRGeno, 2019-10-13:
I clicked on the Full browser mode and don't seem to be able to get it to go back to integrated browser mode.

StefanCohen, 2019-10-13:
You get out of it by clicking on the full browser icon again or by pressing the back button in the browser. I have tested this in the latest version of Chrome, Firefox and Safari on Mac. What browser/OS are you using?

PRGeno, 2019-10-13:
Latest version of Chrome.  The back button brings me back to the previous tab view, but clicking on the dashboard tab goes right back to Full Browser mode, not integrated mode.  Clicking the full browser button again doesn't reverse the setting for me.  

The only way I was able to get it back to normal integrated mode was to manually type in the ipaddress/#tab_plugin_dashboard address, removing the /?dashboard=full portion of the address.

StefanCohen, 2019-10-13:
Pretty strange. I can switch seamlessly between the integrated, full browser and full screen modes using the icons, esc and the browser back button in all three browsers i mentioned above.

PRGeno, 2019-10-13:
That is odd.  Esc works as expected for full screen mode, but nothing works for me to reset full browser mode back to integrated mode except manually removing the ?dashboard=full portion of the address.

StefanCohen, 2019-10-13:
Can you try emptying the browser cache?

StefanCohen, 2019-10-13:
I just realized that there probably is one thing I can do to force the browser to behave when pressing the icon again to get out of full browser mode. It is getting late here but I will give that a go tomorrow morning.

PRGeno, 2019-10-14:
Clearing the cache unfortunately doesn't change the behavior for me.

StefanCohen, 2019-10-14:
Try this. Change Line 73 in dashboard.js to:

`location.href="/";`

Then reload the page and try switching to and back from full browser mode using the icon.

PRGeno, 2019-10-14:
In the interest in providing as much detail as I can here is some more info about my set up.  

I'm running OctoPrint on an Ubuntu 19.04 installation on an Intel box.  I've installed all plugins through the plugin manager so they are stored as packaged plugins in the packed_plugins.js.  

I changed the line you referenced as you described but it did not change the behavior even after restarting OctoPrint.  

I've tested this on 2 different computers with both Chrome and Firefox on each.  The behavior I describe is consistent and the only way I can get rid of full browser mode is to manually delete "?dashboard=full" from the address bar.

This really isn't a huge problem for me, but I assume you would prefer it work as you intended.  I hope this provides some help in troubleshooting

StefanCohen, 2019-10-14:
I appreciate it and it does provide valuable feedback. My problem right now is that I can't reproduce it no matter how hard I try. 

I have three separate RPis. One for development, one for testing and one for my printer. I also have a a machine running Ubuntu 18.04 that I do OctoPrint testing on in addition to the RPIs. I have five different browsers across both win and osx machines and no combination of host/client shows the problem as you describe it so far.

The line change you did *should* remove that "?dashboard=full" from the address bar for you and I'm a bit concerned that it does not. It points to two possibilities:

Either the change is not applied, or something is caching the requests. 

Do you by any chance run OctoPrint behind some kind of proxy server? That could explain the problem.

PRGeno, 2019-10-14:
No proxy.  I do some software development too and can understand the dilemma of not being able to reproduce a problem to troubleshoot it.  I really don't have a very unique setup other than running it on an Intel based Ubuntu platform.   

Based upon that, I can't understand how I'd be the only one experiencing this issue, but if I am it isn't that big of a deal for me and really doesn't hamper my use of Dashboard much at all.

I'll try to dig a little deeper when I have a chance and see if I can come up with any other tidbits that might help identify why this is happening.  I have a second, identical OctoPrint box connected to another Prusa Bear Mk3s.  It has not been updated with your latest release because it's in the middle of a rather long print.  I'll see if it reacts differently, or the same, once that prints finishes (currently slated to finish late tonight).

StefanCohen, 2019-10-14:
There is now at least one other user who has this problem so you are not alone. I have made some changes that may fix this for you but I can't verify it myself. The changes are not in any release yet so please try it by installing via the plugin manager directly from this URL: https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

PRGeno, 2019-10-14:
I installed your update from the link provided.  Dashboard does now go back to the regular integrated mode as would be expected in both Chrome and Firefox. It appears you found the issue, at least in my case.

One other thing with Full Browser mode.  When in integrated browser mode clicking on the full browser button flashes a black screen with a "Please wait" message then goes back the the integrated view.  Hitting the Full Browser mode button again a second time will then open Full Browser mode.  It always takes 2 hits of the button to get it to eventually go into that mode, with the black plaease wait screen in between.  This is with both Chrome and Firefox, and is the case with both this pre-released fixed version as well as the previous released version.

StefanCohen, 2019-10-14:
Ok. I'm glad that fixed at least that issue but again, that second behaviour is not something I see here. It will always go directly to the selected mode for me. I will dig some more...

Btw, are you using a different tab order than default?

PRGeno, 2019-10-14:
Good point, yes I am using the Tab Order plugin  to create a custom tab order (see attached)
[image]

StefanCohen, 2019-10-15:
Thanks. I will have to install Tab Order again and see if that is causing the problem.

closed 2019-10-21 by StefanCohen