j7126/OctoPrint-Dashboard #97: 1.11.0 doesn't respect url for full browser modr

ochm, 2019-10-14:
I'm running two instances of Octoprint for two printers.
http://xx.xx.xx.xx/P1/
http://xx.xx.xx.xx/P2/

Everything seems to works well, but Full browser window returns
503 Service Unavailable
No server is available to handle this request. 

And it is because it is trying to open http://xx.xx.xx.xx/?dashboard=full
instead of url begin as described above.

**Expected behavior**
Respect url where the page is opened

Platform: RPi 4
OctoPrint version : 1.3.11
OctoPi version    : 0.17.0

 - OS and version: W8.1 64bit
 - Browser any you like

StefanCohen, 2019-10-14:
Thanks. I see the problem and will work on a fix.

StefanCohen, 2019-10-14:
I have a fix here that I hope will fix this issue for you but I can't verify it locally. Please test it by install directly from this URL: Please try it by installing via the plugin manager directly from this URL: https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

StefanCohen, 2019-10-21:
Attempted fix in 1.11.1

closed 2019-10-21 by StefanCohen