j7126/OctoPrint-Dashboard #98: Bad formatting in Dashboard 1.11.0 plugin settings

ochm, 2019-10-14:
**Describe the bug**
Bad formatting of input boxes and texts

**To Reproduce**
Open dashboard plugin settings

**Expected behavior**
Proper formatting

**Screenshots**
[image]

[image]

Platform: RPi 4
OctoPrint version : 1.3.11
OctoPi version    : 0.17.0

**Desktop (please complete the following information):**
OS and version: W8.1 64bit
Chrome, Firefox

StefanCohen, 2019-10-14:
Interesting. It doesn’t render like that for me in any browser.

StefanCohen, 2019-10-14:
This is how it looks here in Safari, Chrome and Firefox:

[image]

I have regardless added some vertical space in the settings dialog and pushed a change (no release yet). Please try it by installing via the plugin manager directly from this URL: https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

ochm, 2019-10-15:
I probably know where is problem. My white theme is default Octoprint theme. Here is bad formatting. Your dark theme is from Themeify plugin (i guess), and yes when i turn on this plugin, the rendering is OK. Only the up and down buttons are very, very small.
[image]

StefanCohen, 2019-10-15:
This is my dashboard in Chrome without Themeify installed:

[image]

This is very strange. What other plugins are you using?

StefanCohen, 2019-10-15:
I guess one method to pinpoint this would be to disable them all except the dashboard to see if it clears the issue and then enable them one by one until it happens again. That would take a while... I will have to postpone that a bit since I'm currently focusing on the OctoPrint 1.4 support.

ochm, 2019-10-16:
Hi, 

I disabled all available plugins, and the problem disapear. 
Then i sequentialy enabled and restarted Octoprint after each enabled plugin and the problem doesn't come back, but only in Chrome. 
In Firefox ESR is the rendering still bad in default theme, even if the browser cache is cleared, 
with Themeify everything looks well. 

I can live with this. :) 

​​-- 
Martin Och 
Mikrotik: MTCNA, MTCWE, MTCTCE 
Veeam: VMTSP, VMSP 
VMware: VCP6-DCV 

BQ_BEGIN

BQ_END

StefanCohen, 2019-10-17:
Ok. I can't reproduce it at all so I think we have to close the issue. At least for now. Please re-open if you find any more info.

ochm, 2019-10-21:
I made short investigation, in Chrome and Firefox ESR to see how the settings of Dashboard are formated..

Without Themeify plugin:
Chrome (latest) OK
Firefox ESR (latest) NOK

With Themeify plugin:
Chrome (latest):
Discorded OK
Material UI Light OK
Cyborg NOK
Discoranged OK
Dyl OK
Nighttime OK

Firefox ESR (latest):
Discorded OK
Material UI Light NOK
Cyborg NOK
Discoranged OK
Dyl OK
Nighttime OK

StefanCohen, 2019-10-21:
Thanks!

I can reproduce the problem now and will try to come up with a fix.

StefanCohen, 2019-10-21:
Ok. This is probably a bug in some of the themeify themes but I have managed to work around it in FF. These are my results:

FF, Material UI Light:
[image]

FF, Cyborg:
[image]

Chrome still exhibits the same problem after the fix so I think you will need to address this with the Themeify plugin author to get a resolution.

closed 2020-01-30 by StefanCohen