j7126/OctoPrint-Dashboard #99: First layer duration is not accurate

purist69, 2019-10-16:
I am using your plugin with a Prusa MK3s.  My first layer is always 2000+ seconds.  It appears (I'm assuming) that the time to heat the extruder and the bed is included in the first layer times.  Please resolve if possible, it skews the graph.

[image]

StefanCohen, 2019-10-16:
Hi and thanks for the feedback. 

That seems odd. 41 minutes for the first layer and then 21 minutes per layer after that? Can you please upload the gcode file here so I can take a look?

purist69, 2019-10-17:
Here's a sample gcode I just created, the others I've been working with are too large.

`; generated by PrusaSlicer 2.1.0+win64 on 2019-10-17 at 00:00:10 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R6
M73 Q0 S6
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.7.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
Tx
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

G21 ; set units to millimeters

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
Tc
; purge line
G1 X55.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0 F2200.0
M73 Q1 S6
M73 P1 R6
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
G1 X5.0 E4.0 F1000.0

M221 S95
G92 E0.0
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K30 ; Filament gcode
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-0.80000 F2100.00000
G1 Z0.600 F10800.000
;AFTER_LAYER_CHANGE
;0.2
G1 X118.844 Y94.608
G1 Z0.200
G1 E0.80000 F2100.00000
M204 S1000
G1 F1064
[... 3080 lines cut ...]
; threads = 8
; toolchange_gcode = 
; top_fill_pattern = rectilinear
; top_infill_extrusion_width = 0.4
; top_solid_infill_speed = 40
; top_solid_layers = 5
; travel_speed = 180
; use_firmware_retraction = 0
; use_relative_e_distances = 1
; use_volumetric_e = 0
; variable_layer_height = 1
; wipe = 1
; wipe_into_infill = 0
; wipe_into_objects = 0
; wipe_tower = 1
; wipe_tower_bridging = 10
; wipe_tower_rotation_angle = 0
; wipe_tower_width = 60
; wipe_tower_x = 170
; wipe_tower_y = 125
; wiping_volumes_extruders = 70,70
; wiping_volumes_matrix = 0
; xy_size_compensation = 0
; z_offset = 0
`

StefanCohen, 2019-10-17:
The layer times are calculated by the DisplayLayerProgress plugin. It does that by finding layer markers (`;BEFORE_LAYER_CHANGE` in Slic3er/PrusaSlicer) and counting the time passed between them when sending the gcodes to the printer. In the example you sent, the heating (`M190 S60 ; wait for bed temp` and `M109 S215 ; wait for extruder temp`) is as you can see done prior to the first layer marker. so the heating time is not included in the first layer time. I tested your gcode example on a virtual printer in OctoPrint and the first layer took about three seconds, so it seems about right and didn't include the heating time which was about five seconds in total on the virtual printer.

I'm still curious about that file that apparently takes 41 minutes on the first layer. Can you please zip it and attach it here or on some separate file sharing site if it is too large?

purist69, 2019-10-17:
Thanks for looking into it.  I'll see where I can post that larger gcode file and also try to print the gcode I shared already and give you a visual output.

purist69, 2019-10-17:
you can get the larger gcode I used here:

https://drive.google.com/file/d/10pTtP05LsMzj2F_XIEMh5IAcbq9NGKPB/view?usp=sharing

purist69, 2019-10-17:
Ok, so not nearly as bad, but as you can tell (30ish seconds), there still a significant time taken to do the first layer.  Every graph looks like a hockey stick from the first layer.  the gcode was modified slightly in that the filament was changed to ABS  (it's what I have loaded) and the infill was changed to 20%.
[image]

purist69, 2019-10-18:
you are correct, my first layer is 20mm/s that must be it.  I assumed better since you said the gcode printed in your virtual printer the first later in 3 seconds.

StefanCohen, 2019-10-18:
Ok. Closing

closed 2019-10-18 by StefanCohen