j7126/OctoPrint-Dashboard #101: Estimated total print time is empty if PrintTimeGenius is not running

mdaneman, 2019-10-22:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
When PrintTimeGenius plugin is not installed/disabled, the "Estimated total print time" value (after the "question mark") is always empty.

**To Reproduce**
Run Dashboard without PrintTimeGenius

**Expected behavior**
Expected the estimated print time shown in the "Approx total print time" box  on the left of Octoprint to be shown here.

**OctoPrint server and Plugins:**
 - Platform: Rpi 3B
 - OS and version: Raspbian Stretch,
 - OctoPrint version: 1.3.11
 - OctoPi version: 
 - Dashboard version 1.11.0
 - DisplayLayerProgress version 1.14.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version Chrome/Safari/any browser

StefanCohen, 2019-10-22:
Thanks for the feedback. I haven't been able to reproduce this at all. I only have PrintTimeGenius installed on one (my "production" machine) out of four OctoPrint servers and none of them shows the problem you describe. 

Here's one job running on my primary dev machine- 1.3.11 (RPi 3b) with Dashboard 1.11.0 and PrintTimeGenius is not installed:

[image]

Here it is with "Show fuzzy time estimates" turned off:

[image]

Do you see any errors in your browser console?

mdaneman, 2019-10-22:
Here's a screenshot showing the issue.  It's possible that the problem is that PrintTimeGenius is actually installed, but the plugin is disabled.  I'll try uninstalling PTG completely and see if it fixes it.

[image]

StefanCohen, 2019-10-22:
I have tried with just disabling the plugin but It still works for me. What does the "Approx. Total Print Time" in the sidebar say when the Dashboard prints "-". 

The thing is that the Dashboard is totally independent from PrintTimeGenius. It gets the value directly from Octoprint so I have a hard time to imagine what could cause this bug.

mdaneman, 2019-10-23:
Come to think of it, it's not showing the estimated print time on left side either.  I have the "M73 ETA Override" plugin running.  I wonder it that's what's doing it.  I'll play around with disabling it and see if that makes a difference.
[image]

StefanCohen, 2019-10-23:
That is the problem. There is no value for me to display on the Dashboard.

closed 2019-10-23 by StefanCohen