j7126/OctoPrint-Dashboard #102: Dashboard and Enclosure Plugin

VladimirMalinin, 2019-10-22:
Hi! Amazing plugin!

But i have a problem with enclosure plugin. If it on, Dashboard write me, what i need install DisplayLayerPlugin. When Enclosure Plugin is off, i don't see problem.

Mi octopi log here
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3757106/octoprint.log)

StefanCohen, 2019-10-22:
You will need to install DisplayLayerProgress. It is required for the Dashboard to work.

VladimirMalinin, 2019-10-23:
> You will need to install DisplayLayerProgress. It is required for the Dashboard to work.

Yes, i use DisplayLayerProgress, and it work perfectly with your Dashboard =) The problem begins, if i install Enclosure Plugin. If I turn off this plugin everything works again.

VladimirMalinin, 2019-10-23:
Maybe some special setup is required for Enclosure Plugin?

StefanCohen, 2019-10-24:
Ok. That is strange. I have Enclosure running together with the Dashboard on three different RPis. 

I can't see anything suspicious in the logfile except a bunch of errors from your language pack but that is probably unrelated. Do you see any errors in the browser console if you turn enclosure back on?

VladimirMalinin, 2019-10-25:
No, i'm not see errors.

So, this is my dashboard, enc plugin turn off:
[image]

Then
Enclosure turned on, and DisplayLayerProgress has updated on 1.14.1

Restart
Ooooops... As soon i install new update for display layer progress, octoprint crashed and load on safe mode. 

I reinstall all plugins now - dashboard, enclosure, display:
First DisplayLayerProgress (1.14.1) - ok
[image]
Second Dashboard (1.11.1) - oooops...  
[image]
Newermind, install Enclosure 4.13. Without changes.

It seems that the plugin settings remained after uninstall. It might be worth trying to remove them, but I don’t know where they are located.

I can give you access to my printer if you want =)
If we don’t find a solution, I’ll try to completely reinstall the octoprint, but this is too radical a way to solve the problem =D

StefanCohen, 2019-10-27:
I have tried to reproduce this multiple times not but I have failed. I will close this for now. Please let me know if it behaves the same on a clean install.

closed 2019-10-27 by StefanCohen

VladimirMalinin, 2019-11-27:
Hi. I reinstall my octoprint on new sd card. install display layer progress, enclosure plugin, dashboard, and he doesn't work. i don't understand what's hAppend.
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3896533/octoprint.log)