j7126/OctoPrint-Dashboard #103: DashboardViewModel missing gcodeViewModel, enclosureViewModel

RedKrieg, 2019-10-22:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
Javascript errors relating to DashboardViewModel missing gcodeViewModel, enclosureViewModel result in the dashboard tab not displaying correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Open http://octopi.local/#tab_plugin_dashboard
2. Open console to see JS errors
3. Dashboard never finishes loading

**Expected behavior**
The dashboard should load and start printing information

**Screenshots**
[image]

**Files (please attach the following Files or screenshots when applicable):**
Browser console log.
```
5A cookie associated with a cross-site resource at <URL> was set without the `SameSite` attribute. A future release of Chrome will only deliver cookies with cross-site requests if they are set with `SameSite=None` and `Secure`. You can review cookies in developer tools under Application>Storage>Cookies and see more details at <URL> and <URL>.
packed_core.js?2550febe:13613 Starting dependency resolution...
packed_core.js?2550febe:13712 Could not instantiate the following view models due to unresolvable dependencies:
(anonymous) @ packed_core.js?2550febe:13712
mightThrow @ packed_libs.js?ef1e2987:3535
process @ packed_libs.js?ef1e2987:3603
setTimeout (async)
(anonymous) @ packed_libs.js?ef1e2987:3641
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
fire @ packed_libs.js?ef1e2987:3407
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
ready @ packed_libs.js?ef1e2987:3879
completed @ packed_libs.js?ef1e2987:3889
packed_core.js?2550febe:13714 dashboardViewModel (missing: gcodeViewModel, enclosureViewModel )
(anonymous) @ packed_core.js?2550febe:13714
Pn @ packed_libs.js?ef1e2987:11208
(anonymous) @ packed_libs.js?ef1e2987:11229
(anonymous) @ packed_core.js?2550febe:13713
mightThrow @ packed_libs.js?ef1e2987:3535
process @ packed_libs.js?ef1e2987:3603
setTimeout (async)
(anonymous) @ packed_libs.js?ef1e2987:3641
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
fire @ packed_libs.js?ef1e2987:3407
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
ready @ packed_libs.js?ef1e2987:3879
completed @ packed_libs.js?ef1e2987:3889
packed_core.js?2550febe:13723 ... dependency resolution done
packed_core.js?2550febe:14015 Initial application setup done, connecting to server...
(index):9000 Initialized error tracking
packed_core.js?2550febe:11688 Connected to the server
packed_core.js?2550febe:13983 Finalizing application startup
packed_core.js?2550febe:13864 Going to bind 36 view models...
packed_core.js?2550febe:13917 Did not bind view model AnywhereViewModel to target #wizard_plugin_anywhere since it does not exist
packed_core.js?2550febe:13917 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?2550febe:13917 Did not bind view model ThespaghettidetectiveBetaViewModel to target #wizard_plugin_thespaghettidetective_beta since it does not exist
packed_core.js?2550febe:13917 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?2550febe:13917 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?2550febe:2489 User redkrieg logged in
packed_core.js?2550febe:13956 ... binding done
packed_core.js?2550febe:13968 Application startup complete
```

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Raspbian GNU/Linux 9 (stretch)
 - OctoPrint version: 1.3.12
 - OctoPi version: 0.16.0
 - Dashboard version 1.11.1
 - DisplayLayerProgress version 1.14.0

**Desktop (please complete the following information):**
 - OS and version: Ubuntu 18.04 Desktop
 - Browser and version: chrome 77.0.3865.120

**Additional context**
I had the plugin working previously, I don't know what I changed that might have caused this issue.

StefanCohen, 2019-10-22:
It looks like the gcodeViewModel is missing for some reason. Have you by any chance disabled the GCODE Visualiser in the OctoPrint settings?

You may also have enabled the "Show Temp Sensor Info from Enclosure Plugin (Experimental)" setting in the Dashboard but not installed the Enclosure Plugin. Is that correct?

RedKrieg, 2019-10-22:
I did have the gcode visualizer unchecked in octoprint, thanks.  I had tried enabling it again but it apparently didn't change the javascript to add it until I restarted the server.  I did not have the experimental enclosure plugin enabled, but there are no longer any errors in the console, so that's probably a red herring.  I was having performance issues with the visualizer enabled, so I had it turned off.  I can leave it on for now.

closed 2019-10-22 by RedKrieg