j7126/OctoPrint-Dashboard #104: Small bug - Fan speed shows 1 percent lower than set

ochm, 2019-10-23:
Hi Stefan, i found small bug in fan speed.
In slicer is set 50%, in printer settings i see number 127 (can be 0-255), but your Dashboard shows 49%. Can you do small correction, please?

Sincerely
Martin

StefanCohen, 2019-10-24:
I can reproduce that by setting the speed to 50% in Cura. The fan speed is however not calculated by the Dashboard plugin. It is one of the metrics produced by DisplayLayerProgress and it is simply rendered by the Dashboard:

[image]

I suggest that you report this issue to the author of [DisplayLayerProgress](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues) so he can take a look at it.

closed 2019-10-24 by StefanCohen

ochm, 2019-10-24:
Done https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/91