j7126/OctoPrint-Dashboard #105: MMU2s - averaging of layer duration

huskyte, 2019-10-23:
MMU prints with PrusaSlicer seem to separate supports and actual prints in different layers. Additional side note: Waiting for the MMU counts to the layer duration. 

I would propose to display in the graph a floating average (adjustable in the setup) plus either a non linear y-scale or a cut off at 200% (also ajustable) the overall average (first layer usually is so slow it ruins the chart).

StefanCohen, 2019-10-24:
Thanks for the feedback!

1. Layer processing is done by the [DisplayLayerProgress plugin](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues) and the results are displayed by the Dashboard. I can't tell how MMU gcode works or are handled since I don't have an MMU but I suggest that you ask the plugin developer. 
Note: _If_ PrusaSlicer inserts a layer marker for each MMU color, that may in fact be a bug in PrusaSlicer unless it is a conscious decision for some reason. It would regardless make it hard for DisplayLayerProgress to work unless it can be worked around somehow to figure out what "layers" are in fact one single layer (I suspect that there are other slicer settings that would prevent looking at the layer height from working reliably). Waiting for the MMU _should_ i.m.o count in the layer duration but it is a bit hard to imagine without an actual MMU printer to play with. 

2. A non-linear y scale is an interesting proposal and I have previously thought about adding an average. Another method to deal with the first layer(s) could be a setting for a rolling window so that only the last X layers are drawn on the graph. I suggest that you create a separate issue for this proposal and that we close this one. Do you agree?

Willmac16, 2020-12-27:
Hi there,
I am one of the maintainers for this plugin, and I now have an MMU2S. Once I get it working properly (fingers crossed) I will work on adding proper support for it in the Dashboard V2 backend.