j7126/OctoPrint-Dashboard #106: Dashboard displays don't work on a manual install of OctoPrint

iandouglas, 2019-10-28:
**Describe the bug**
I've installed the OctoPrint Dashboard plugin and several plugins that it says it needs or works well with, but my display doesn't have any information on it. It's been like this for several of the latest versions.

**To Reproduce**
Install the plugins on top of a manual install of OctoPrint

**Expected behavior**
I would expect the dashboard to show graphs of data like the plugin suggests.

**Screenshots**
[image]

**Files (please attach the following Files or screenshots when applicable):**

**OctoPrint server and Plugins:**
Raspberry Pi 4B, 2GB model, though also had this problem on a Pi 3B as well
Raspbian Buster, not OctoPi build since I'm on a Pi 4, though this happened on my Pi 3B using OctoPi 0.13 as well
OctoPrint 1.3.12
Dashboard 1.11.1
DisplayLayerProgress 1.14.2
PrintTimeGenius 2.0.2

**Desktop (please complete the following information):**
- Mac OS Mojave and Windows 10
- Chrome v77

---

My first guess was that maybe there's something fundamental missing from a manual OctoPrint installation versus an OctoPi installation. But since this was also happening on my Pi 3B that had OctoPi installed on it, I ruled that out as part of the problem. I'm happy to pull any stats or information about my Pi4B that you think would be helpful.

StefanCohen, 2019-10-28:
I have several manually installed instances including a 4B (4Gb) that works correctly so this is a bit strange. 

Please attach your octoprint log and also take a look in your browser terminal. Do you have any error messages there?

iandouglas, 2019-10-28:
When I refresh I get the following console errors
[image]

Also when I refresh I got the following octoprint log lines:
```

2019-10-28 02:11:46,216 - octoprint.plugins.palette2 - INFO - Omega: read in line: O97 U26 D00002117
2019-10-28 02:11:46,217 - octoprint.plugins.palette2 - INFO - 8471mm used
2019-10-28 02:11:46,250 - octoprint.plugins.palette2 - INFO - Updating UI: filamentLength
2019-10-28 02:11:47,508 - octoprint.server.util.flask - INFO - Passively logging in user id from ::ffff:192.168.86.249
2019-10-28 02:11:47,557 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.86.249
2019-10-28 02:11:48,327 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.86.249
2019-10-28 02:11:48,396 - octoprint.server.util.flask - INFO - Passively logging in user id from ::ffff:192.168.86.249
2019-10-28 02:11:50,188 - octoprint.plugins.webcamstreamer - INFO - Checking stream status.
2019-10-28 02:11:50,207 - octoprint.plugins.webcamstreamer - INFO - WebStreamer is streaming
2019-10-28 02:11:52,695 - octoprint.plugins.palette2 - INFO - Got a command uiUpdate
2019-10-28 02:11:52,696 - octoprint.plugins.palette2 - INFO - Updating all UI variables
2019-10-28 02:11:52,699 - octoprint.plugins.palette2 - INFO - ADVANCED UPDATE UI
2019-10-28 02:11:52,715 - octoprint.plugins.palette2 - INFO - Got a command uiUpdate
2019-10-28 02:11:52,716 - octoprint.plugins.palette2 - INFO - Updating all UI variables
2019-10-28 02:11:52,718 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,719 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,720 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,721 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,721 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,722 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,723 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,724 - octoprint.plugins.palette2 - INFO - ADVANCED UPDATE UI
2019-10-28 02:11:52,725 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,726 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,727 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,727 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,729 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,730 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,730 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,801 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:52,935 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,021 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,075 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,103 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,131 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,144 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2019-10-28 02:11:53,150 - octoprint.server.util.sockjs - INFO - User id logged in on the socket from client ::ffff:192.168.86.249
2019-10-28 02:11:53,235 - octoprint.plugins.canvas - INFO - Sending UIUpdate from Canvas
2019-10-28 02:11:53,363 - octoprint.plugins.palette2 - INFO - Got a command changeAutoCancelPing
2019-10-28 02:11:53,510 - octoprint.plugins.palette2 - INFO - ADVANCED: autoCancelPing -> 'False' 'False'
2019-10-28 02:11:53,521 - octoprint.plugins.palette2 - INFO - Got a command changeFeedRateSlowPct
2019-10-28 02:11:53,523 - octoprint.plugins.palette2 - INFO - Splice Feed Rate Speed did not change. Do nothing
2019-10-28 02:11:53,534 - octoprint.plugins.palette2 - INFO - Got a command changeShowPingOnPrinter
2019-10-28 02:11:53,662 - octoprint.plugins.palette2 - INFO - ADVANCED: showPingOnPrinter -> 'False' 'False'
2019-10-28 02:11:53,674 - octoprint.plugins.palette2 - INFO - Got a command changeFeedRateControl
[... cut ...] :1570602775}]}},"version":3,"timestamp":1572250313,"clientToken":"3ac9967f-cfab-43f6-914e-bf3b6e1780b2"}
2019-10-28 02:11:53,778 - octoprint.plugins.canvas - INFO - Status: accepted
2019-10-28 02:11:53,780 - octoprint.plugins.canvas - INFO - Token: 3ac9967f-cfab-43f6-914e-bf3b6e1780b2
2019-10-28 02:11:53,781 - octoprint.plugins.canvas - INFO - No delta found in object. No action needed.
2019-10-28 02:11:53,898 - octoprint.plugins.palette2 - INFO - ADVANCED: feedRateControl -> 'False' 'False'
```

iandouglas, 2019-10-28:
I find it odd that per my original screenshot at the top, the Dashboard plugin things that the DisplayLayerProgress plugin isn't installed when it is. What else can I offer to help debug this?

StefanCohen, 2019-10-28:
You see that error message in the Dashboard because the Dashboard viewModel binding fails so early. The error message is part of the html template and it is supposed to be hidden when the displayLayerProgress viewmodel is detected, but it doesn't get that far and so the message is visible. I can try another method to inject it into the page once we have resolved your issue.

The first problem is that the gcodeViewModel is missing. Have you by any chance disabled the gCode Visualizer in the octoprint settings? In that case, try enabling it and restart octoprint to see if that error message disappears.

iandouglas, 2019-10-28:
I did have the GCODE viewer disabled, but re-enabling it still shows me that error. Re-enabling it didn't prompt to restart OctoPrint, but I just started a new 6-hour print, so I'll restart OctoPrint later this afternoon to see if that has any effect. Right now I see nothing different, and I still see that console error about gcodeViewModel.

StefanCohen, 2019-10-28:
That is expected. The binding is only done on startup of the plugin.

iandouglas, 2019-10-29:
Re-enabling the gcode viewer worked. Maybe add that to the docs?

closed 2019-10-29 by iandouglas

StefanCohen, 2019-10-29:
That dependency is described in the [dependencies section in the readme](https://github.com/StefanCohen/OctoPrint-Dashboard#dependencies). I think I will have to try to detect the presence of that viewmodel programatically and display an error message.

lloydbigoil, 2021-07-06:
i have this same issue on an iPad trying to run the dashboard via local network

Willmac16, 2021-07-06:
@lloydbigoil If you have the same visual appearance that means that there was a knockout binding issue preventing the whole Dashboard tab from rendering. If you could get us a browser log (I believe by plugging into a Mac with usb for iPadOS Safari), we would be able to debug the issue.

lloydbigoil, 2021-07-06:
damn, pc only here. same problem on iphone 6 safari too, wanted to use one as my browser.

Willmac16, 2021-07-06:
@lloydbigoil does the issue pop up on pc? I know some code that doesn't play nice with older iPads snuck into octoprint a while ago. If you can, update octoprint.

lloydbigoil, 2021-07-06:
Fully up to date. ran all update via the gui then  ssh and ran update and upgrade. only downloaded the plugin recently from the octoprint plugin store.

lloydbigoil, 2021-07-06:
also no my desktop running windows 10 or ubuntu 20 is fine

Willmac16, 2021-07-06:
@lloydbigoil What gen IPad is it?

j7126, 2021-07-06:
@lloydbigoil if you are running the latest version of dashboard (1.18.3) it _should_ log the fontend errors to octoprint.log so you could try to upload your octoprint.log. The new logging feature has not been tested yet to may or may not work.

lloydbigoil, 2021-07-06:
> @lloydbigoil What gen IPad is it?

4th Gen

> @lloydbigoil What gen IPad is it?

4th gen.

> @lloydbigoil if you are running the latest version of dashboard (1.18.3) it _should_ log the fontend errors to octoprint.log so you could try to upload your octoprint.log. The new logging feature has not been tested yet to may or may not work.

i checked the log, then made sure to reproduce the error on the ipad. the log only shows this https://pastebin.com/AZwzscQK for my ipad connecting and then disconect. confirmed this is the iPad looking at the ip address in general