j7126/OctoPrint-Dashboard #109: Cannot get stats from DisplayLayerProgress even though it is installed.

ilovecats516, 2019-11-17:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
Cannot get stats from DisplayLayerProgress even though it is installed.

**To Reproduce**
Install Dashboard.
Install DisplayLayerProgress.

**Expected behaviour**
Stats show up on the dashboard.

**Files (please attach the following Files or screenshots when applicable):**
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3856238/octoprint.log)

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OctoPrint version: 1.3.12
 - OctoPi version: 0.16.0
 - Dashboard version 1.11.1
 - DisplayLayerProgress version 1.14.2

**Desktop (please complete the following information):**
 - OS and version: MacOS Catalina
 - Browser and version: Safari 13.0.3

StefanCohen, 2019-11-18:
Hi. I can't see anything suspicious in your log and I have it working with the exact same setup as you do. Do you have any errors in your browser console?

StefanCohen, 2019-11-20:
Closing due to lack of response. Please feel free to re-open if needed.

closed 2019-11-20 by StefanCohen

ilovecats516, 2019-12-07:
Sorry for the slow response. Yes, there are errors in the console (although some of them are Grammarly)
[image]