j7126/OctoPrint-Dashboard #110: Layer Progress always displays 0% on touch UI

RyanParle, 2019-11-20:
**Describe the bug**
Layer Progress always displays 0% on touch UI

**To Reproduce**
Go to Ocotprint-Dashboard tab on Touch UI

**Expected behavior**
To see layer progress bar show how far through current layer the print is.

**OctoPrint server and Plugins:**
 - Platform: RPi 4b 4GB
 - OS and version: Raspbian (i think)
 - OctoPrint version: OctoPrint 1.3.12
 - OctoPi version: OctoPi 0.17.0
 - Dashboard version Dashboard 1.11.1
 - DisplayLayerProgress version 1.14.2
- TouchUI Version 0.3.13

**Desktop :**
 - OS and version: Windows 10
 - Browser and version: Opera 64.0.3417.92

**Touch:**
Huawei Mate 20 Pro / Opera 54.3.26.72.50220
7" Elecrow HDMI touch screen on RPi 4b 4GB / Touch UI Version 0.3.13

StefanCohen, 2019-11-20:
I don't use TouchUI so it is hard to tell but it may have disabled or altered the Gcode Viewer tab so that the layer progress can't be retreived. Could that be it?

MikeUSNRet, 2019-11-24:
On my ender 3, both on the system readout and on Octoprint Dashboard, for Layer, I am only getting job completion percent and L= -/- so it looks like this on my current print:  17% L= -/-

OctoPrint Server and Plugins:
PRi 3b+ 4GB
OctoPrint Version: Octoprint 1.3.12
OctoPi Version: 0.16.0
Dashboard Version: 1.11.1
DisplayLayerProgress Version: 1.14.2

Cura Slicer :  4.3

So the question is, where and how do I actually get the Layer information to display correctly?

StefanCohen, 2019-11-24:
@MikeUSNRet That seems unrelated to the issue @RyanParle reported. Your problem is likely related to DisplayLayerProgress since it doesn't seem to be able to report the correct layer height to Dashboard. I suggest that you report it to the maintainer of that plugin.

closed 2019-12-08 by StefanCohen

BillyBlaze, 2020-03-08:
> I don't use TouchUI so it is hard to tell but it may have disabled or altered the Gcode Viewer tab so that the layer progress can't be retreived. Could that be it?

Indeed, the GCode Viewer tab is completely disabled in TouchUI when using 'touchscreen mode'; this is to make performance better. I can confirm that it does work in TouchUI when touchscreen mode is disabled.

StefanCohen, 2020-03-08:
Thanks. The help in the Wiki has been updated with this info.