j7126/OctoPrint-Dashboard #111: Link Enclosure Temperature to "Chamber Temp" guage

ainuke, 2019-11-20:
Not a problem, just an enhancement, I suppose. There may already be a way to achieve this that I'm not seeing.

I've just enabled the Enclosure plugin features in Dashboard, and it appears to be working well reading the temperature of my enclosure and displaying it in a widget. Is there a way to have these temps (current and target) use the "Chamber" gauge, similar to how the hotend and build plate do, for this reading? I'm currently using Enclosure Plugin to maintain the temperature within the enclosure through GIPO.

Ideally, I would see temperature current/target on one gauge, and humidity current/target on a second (although I don't have a dehumidifier attached, so humidity set point not an issue). In a perfect world, there would be a notification pushed (SMS, etc.) if current is not meeting target, but I think that's a request for the Enclosure Plugin author, so disregard.

Thank you for an awesome plugin! It works great and looks great full screen on my second monitor. I certainly get eye rolls from my wife for it, but about as many as I do for 3D printing useful household items, LOL.

Erik

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Ideally in form of a user story so that I know why this is requested:

Example: As a Dasboard user, I want to [feature], because [reason]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

StefanCohen, 2019-11-20:
Thanks for the feedback. There is currently no way to display the Enclosure Plugin sensor readings as temp gauges. One part of the problem is that those sensors doesn't have a target temp, only a current temp. There is also the problem that one sensor may present multiple values (temp + humidity), but there is no way to query the sensor type to determine what to expect.

ainuke, 2019-11-20:
OK; I guess that settles it. Would've been nice, though.
Thank you for your response.
Erik

closed 2019-11-20 by ainuke