j7126/OctoPrint-Dashboard #113: Clean up UI

tablatronix, 2019-12-04:
The UI leaves a bit to be desired.

Ill look and see what this is written in and try to PR some of it, unless someone already has a fork?

- [ ] hide excessive warnings, `Warning! Can't get stats from DisplayLayerProgress. Is it installed` optional or if layer progress is not enabled in settings stop showing this all the time..

- [ ] Hide comm error that show up in the interface now and then

- [ ] Hide ALL interface on load via css, then show enabled items, its pretty ugly to see every element show up when loading and then have it all resize and reflow

- [ ] replace the center line guage indicators (target) with crossing level lines instead maybe, just looks bad and the contrast is confusing

- [ ] minor spacing adjustments, the buttons need more space etc

- [ ] how to change theme colors? Terrible contrast on dark theme

StefanCohen, 2019-12-08:
Thanks for the feedback. Feel free to provide a PR.

tablatronix, 2019-12-09:
> DisplayLayerProgress is a requirement. The plugin will not work without it being installed and enabled.

Why is that? I use it without it, it works fine enough,  when I had it(displaylayerprogress) enabled It messed up my ender 3 screen, and I saw no way to disable get layer and display layer functionality in it.

>The Dashboard doesn't to my knowledge produce any comm errors. Those are likely coming from OctoPrint or another plugin.

I have been trying to reproduce , ill post a screen shot

> Theme colors are changed using Themeify as per the documentation.
I have themify, it does not use themes colors automatically? You MUST use custom? Is there not a way to inherit global classes from themify?

https://github.com/StefanCohen/OctoPrint-Dashboard/wiki/Help just says the class names, no other information so I assumed this was supported by default

tablatronix, 2019-12-09:
[image]

StefanCohen, 2019-12-09:
Thanks for the updates.

The following metrics are provided by DisplayLayerProgress:

- Fan Speed
- Layer Durations
- Current Layer Height
- Total Height
- Current Layer
- Total Number of layers
- Layer average

Plenty of users install the Dashboard without reading the installation instructions and post issues here when the above metrics are not displayed. Hence the warning in the UI.

That comm error is coming from OctoPrint core and has nothing to do with Dashboard except that it it displayed there in addition to the sidebar and in the log. You will need to address the underlying issue in your installation for the error message to go away.

I haven't found a way to inherit theme colors from Themeify yet so custom selectors is the way to go. Feel free to provide a PR if you have a method that works across all Themeify themes.

tablatronix, 2019-12-09:
That comm error is in a notification box, it is not something that gets output like that, It only shows up in the dashboard, broken html tags maybe? My point is it should not be showing there.

Yeah that is why I suggested hiding the warnings when you disable those meters in the settings. I see why it is useful to you for support, but dashboard works fine without displayerlayerprogress for the other stuff. (displaylayerprogress is buggy), so why not make it optional.

StefanCohen, 2020-01-07:
That comm error is provided by the OctoPrint API when querying the printer status. 

I disagree. DisplayLayerProgres is not optional. It is required as described in the Readme since about half of the metrics are provided by that plugin.

tablatronix, 2020-01-07:
The comm error shows in the middle of the ui thats my point.

closed 2020-01-30 by StefanCohen

firozdang, 2021-01-31:
I have this same issue, I installed this multiple times but cant find the  DisplayLayerProgress to clean or do some settings, how do I fix this?

Willmac16, 2021-01-31:
@firozdang
To install DisplayLayerProgress use the plugin manager and install it.

or

In the current version of Dashboard go to the Dashboard settings section, then select the `other settings` tab, then check the enable box for `Supress DLP warning messages`.

firozdang, 2021-01-31:
worked, please close

tablatronix, 2021-01-31:
I have had this disabled, I will update and try it again now