j7126/OctoPrint-Dashboard #114: Reopening - Cannot get stats from DisplayLayerProgress even though it is installed.

ilovecats516, 2019-12-07:
Describe the bug
Cannot get stats from DisplayLayerProgress even though it is installed.

To Reproduce
Install Dashboard.
Install DisplayLayerProgress.

Expected behaviour
Stats show up on the dashboard.

Files (please attach the following Files or screenshots when applicable):
[octoprint-2.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3935644/octoprint-2.log)

OctoPrint server and Plugins:

Platform: RPi 3b+
OctoPrint version: 1.3.12
OctoPi version: 0.16.0
Dashboard version 1.11.1
DisplayLayerProgress version 1.14.2
Desktop (please complete the following information):

OS and version: MacOS Catalina
Browser and version: Safari 13.0.3

ilovecats516, 2019-12-07:
Browser console: 
[image]

StefanCohen, 2019-12-08:
The browser log provides a few clues. It seems like you have disabled the gcode viewer in the OctoPrint settings. That viewmodel is required by my plugin. It also seems like you have enabled a sensor (Show Temp Sensor Info from Enclosure Plugin (Experimental)) in the dashboard settings because it is also complaining about the enclosure viewmodel. Please enable the gcode viewer and disable the sensor view (unless you have the enclosure plugin installed and configured)

ilovecats516, 2019-12-10:
Enabling the GCode viewer fixed it, thanks. I didn't have the Show Temp Sensor Info from Enclosure Plugin box checked though, but all is well. Thanks again.

closed 2019-12-10 by ilovecats516