j7126/OctoPrint-Dashboard #116: Dashboard causing layer shift on Prusa MK3S

mdaneman, 2020-01-02:
**Describe the bug**
When printing through Octoprint with Dashboard plugin enabled, I get X-axis layer shifts of ~0.6mm on nearly every print, often with multiple shifts per print. The layer shift is such that once a layer shifts, all layers above it are shifted by the same amount.

**To Reproduce**
I confirmed that the layer shifts DO NOT occur under the following conditions:
1. Printing from SD card (no Octoprint)
2. Printing from Octoprint with Dashboard plugin disabled.  I do still have the folloiwng plugins enabled: DeleteAfterPrint, DisplayLayerProgress, "M73 ETA Override", Themeify, TouchUI.

Only when I enable Dashboard do the layer shifts start.

Additionally, I have been running Dashboard for a few months with no problems before the layer shifts started, so I suspect that the issue is related to upgrading to one of the two most recent versions of Dashboard.  I'd like to try revering to an older version, but don't know how to do that.

**Expected behavior**
No layer shifts when running with Dashboard.

**Files (please attach the following Files or screenshots when applicable):**
The layer shifts happen with any gcode file.

**OctoPrint server and Plugins:**
 - Platform: RPi 3b
 - OS and version: Raspbian Stretch
 - OctoPrint version: OctoPrint 1.3.12
 - Dashboard version: Dashboard 1.11.1
 - DisplayLayerProgress version: 1.15.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version: independent of browser, but I usually use chrome

**Additional context**
I have very high confidence that the problem is related to Dashboard, since I can make the layer shifts appear or disappear by enabling/disabling Dashboard plugin.  Although I suppose it could be Dashboard interacting with another plugin.

I would really like to try reverting to an older version of Dashboard (maybe 2 versions back) to see if the problem goes away, but don't know how to do that in Octoprint.  Would appreciate advice on how to do that.

j7126, 2020-01-03:
You can downgrade to an older version by first removing the plugin and then pressing get more and entering
https://github.com/StefanCohen/OctoPrint-Dashboard/archive/[VERSION_NAME_HERE].zip
into install from url. 
for example
https://github.com/StefanCohen/OctoPrint-Dashboard/archive/1.10.2.zip
for 1.10.2
Please advise if this causes the problem to stop

mdaneman, 2020-01-03:
Great!  Thanks a lot for your help.  I'll give it a try.

mdaneman, 2020-01-04:
Looks like I spoke too soon.  I disabled Dashboard and still got layer shifts printing the same model.  I then printed from SD and it printed fine.  So it looks like it's not necessarily Dashboard related, but it is Octoprint related.  It's funny because I've been printing through Ocotprint without Dashboard for about a month and had zero layer shifts with other models. You can close this issue for now.

closed 2020-01-07 by StefanCohen