j7126/OctoPrint-Dashboard #117: Missing layer time chart

ochm, 2020-01-05:
Ia far as i know, i did only update DisplayLayerProgress Plugin: to 1.16.0,
and from this time the dashboard doesn't show some informations
[image]
Should i do downgrade the Display layer progress plugin, or is some chance to repair Dasboard?
Sincerely

divisl, 2020-01-05:
I have the same problém, thx.

ochm, 2020-01-05:
Probably bug in Plugin as described in https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/104

StefanCohen, 2020-01-07:
This is most likely the bug in DisplayLayerProgress as you mentioned as the latest version doesn't provide the layer updates as previous versions did. Closing.

closed 2020-01-07 by StefanCohen