j7126/OctoPrint-Dashboard #118: Total Height Not Displaying in Dashboard

nolspeed29, 2020-01-13:
I am successfully getting the current/max layer number and height parsed by the DisplayLayerProgress plugin.  However, in the Dashboard, I am only getting the correct behavior from layers, while the height shows the correct current height but not total height.  See here:

[image]

I am slicing using PrusaSlicer-2.1.1, and use the following "Before Layer Change" custom G-Code:

`;BEFORE_LAYER_CHANGE
;Z:[layer_z]`

In DisplayLayerProgress, I use the default layer indicator code (which counts the # of ";BEFORE_LAYER_CHANGE" comments in the code.  I could not get the max z height to work unless I use the custom option to get max Z value from an expression, using this expression:

`;Z:([0-9]+[.]*[0-9]*).*`

The default option to get the "Max Z-Value in G-Code", would pull a Z value from the startup/finish GCODE, always setting my max to that high value.  If I used the "Max Z-Value with Extrusion on Z-Layer" option, I would get '0.0' for my max, I think due to the way that the GCODE moves are created by the slicer.

Attached is the GCODE used to recreate this issue and octoprint.log.

[kallax-vinyl-spacer_Mini2_PLA_Standard_No_Supports_PolyPlus_PLA_195.gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4054353/kallax-vinyl-spacer_Mini2_PLA_Standard_No_Supports_PolyPlus_PLA_195.gcode.zip)
 
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4054390/octoprint.log)

**OctoPrint server and Plugins:**
model: Raspberry Pi 3 Model B Plus Rev 1.3
octopi_version: 0.15.1
Octoprint Version: 1.3.12
python: pip: 9.0.3, version: 2.7.13
Dashboard: 1.11.1
DisplayLayerProgress: 1.16.0
Browser: Chrome 79.0.3945.88

Additionally, I am running the Detailed Progress Plugin (0.1.4), to get better info on my printers LCD.  

Any help would be much appreciated!

AndrewStick, 2020-01-17:
[image]
I have the same issue

CallistoJaro, 2020-01-21:
Same issue like AndrewStick
Total height shown on the left side bar and on the top nav bar but not in the dashboard??

Best regards

ochm, 2020-01-22:
It is known issue with DisplayLayerProgress 1.16.0, in 1.15.0 is everything OK.
Unofrunately, in 1.17.0 is the problem similar
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/109

StefanCohen, 2020-01-30:
Thanks for the feedback and sorry for the late response. I've had other things to take care of for a while. I'm keeping this issue open for visibility until Olli has a patch available.

OllisGit, 2020-01-30:
Hi,
the "Total Height" bug is fixed in DLP Version 1.17.1

BR
Olli

AndrewStick, 2020-02-05:
Now it works. Thank you!

closed 2020-02-05 by StefanCohen