j7126/OctoPrint-Dashboard #119: Would love to be able to toggle webcam on and off from dashboard

brandonscholet, 2020-01-15:
First off, I just wanted to say I love this project. Best Plugin Ever!

I love to leave my dashboard up on my computer and phone and just let it go!

I do think it consumes too much bandwidth on my mobile though. I would love to be able to have a way to easy toggle the webcam stream right from the dashboard

click to disappear, click to open back up.

What do you think?

CynanX, 2020-05-18:
It'd also be great if we could switch between webcams (support for MultiCam) So a toggle like, Off | Cam1 | Cam2 etc.

StefanCohen, 2020-05-18:
@CynanX Please see: #65 and #140 I'd love a PR for multicam support since this is an area I've been struggling with for a long while without success

CynanX, 2020-05-19:
Ok I've had a go! I've managed to pop some buttons above the webcam image (Off, Camera 1, Camera 2 (if new URL populated in the settings - I could not see a way to integrate with MultiCam as that does not provide any helpers that I could see) and it seems to work. I'm sure it could use a tidy up with someone that knows better than me... however I can't seem to create a pull request. I tried to push a new branch up and received access denied.

CynanX, 2020-05-19:
Figured it out! New pull request open.

StefanCohen, 2020-05-20:
Thanks. I'll check it out.

StefanCohen, 2020-05-20:
Reviewed. I found a few issues and have a couple of questions. Please see comments in the PR.

CynanX, 2020-05-21:
New draft PR up with comments. Needs some work but could use some help.

CynanX, 2020-05-28:
This has now been merged and will be in the next release !

StefanCohen, 2020-06-02:
The patch from CynanX will be included in 1.13.0.

StefanCohen, 2020-06-02:
Released now. It will show up in OctoPrint in a while when the repos are synced.

closed 2020-06-02 by StefanCohen