j7126/OctoPrint-Dashboard #120: Webcam does not work, then i use a dyndns adress

dr-apple, 2020-01-26:
Hi with this adress http://192.168.0.112:8080/?action=stream in my webcam settings in octoprint works the webcam in the dashboard, but then i use this adress : http://drapple.no-ip.org:8000/++video?cameraNum=18&width=1280&height=720&89706 ( this is my external cam server) it works in octoprint, but not in the dashboard, why ?

StefanCohen, 2020-01-30:
Hmm. The webcam integration has always been a struggle. I will need to debug this but I will first have to set up a new dev env as I have temporarily re-purposed the old one so it may take a while.

StefanCohen, 2020-03-04:
I have looked in to this a bit more. I have no problems with any public web cam feed that I've tried so far. (ex: http://217.152.196.254/axis-cgi/mjpg/video.cgi?camera=1& )

The URL is returned by the embedURL function like this:

return self.settingsViewModel.settings.webcam.streamUrl() + '?' + new Date().getTime();

I added the last part some releases ago to fix an issue with browsers stopping the feed after a while. Could that cause a problem for your video feed server?

StefanCohen, 2020-03-05:
Also, That link you provided requires a login for external access. I'm not sure how that authentication is handled by OctoPrint. Does it use a cookie or some kind of session token after a successful auth?

StefanCohen, 2020-03-06:
I'm closing this for now. Please re-open if the proplem persists.

closed 2020-03-06 by StefanCohen

deennoo, 2020-04-17:
> return self.settingsViewModel.settings.webcam.streamUrl();

this trick work for me.

Video url is : http://ip/videostream.cgi?user=........&pwd=.............

then i have a static picture