j7126/OctoPrint-Dashboard #122: Estimated total print time is empty while printing via API

ochm, 2020-01-27:
I found problem with Estimated total print time in Dashboard (1.11.1)
The time is displayed only when the gcode is printed from internal storage.
When the print is sent via API (from Prusa Slicer), the Estimated total print time is empty.
Information from author of DisplayLayerPlugin - this item is not driven by his plugin.
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/109#issuecomment-578530574
Thank you.

StefanCohen, 2020-01-30:
The estimated total print time is coming straight from OctoPrint via the PrinterStateModel viewmodel and is not manipulated in any way by the Dashboard. Like this:

`printerStateModel.estimatedPrintTimeString()`

I'm regularly printing via the API (from Cura) and haven't seen a problem with this yet. I haven't tried with prusaslicer.

closed 2020-01-30 by StefanCohen

ochm, 2020-01-30:
So the problem is where?

ochm, 2020-02-01:
I tried open issue in Octoprint https://github.com/foosel/OctoPrint/issues/3439