j7126/OctoPrint-Dashboard #123: Finish time is one hour off (timezone??)

AlbanT, 2020-01-30:
Hi,

I noticed that the estimated time that my print is finished is off by one hour. My first thought is timezone as I live in GMT+1. So I expected to fix it by SSH into my Raspberry and use raspi-config to change to the correct timezone (from London to Amsterdam) but this didn't fix my issue :(
Any ideas on where octoprint or this dashboard gets its local time from?
[image]

StefanCohen, 2020-01-30:
Hi. This is from memory as I have temporarily re-purposed my main dev RPi. The ETA is calculated by the Dashboard by adding time remaining to the current time returned by the system. The code:

```
 self.getEta = function (seconds) {
            dt = new Date();
            dt.setSeconds(dt.getSeconds() + seconds)
            return dt.toTimeString().split(' ')[0];
        };
```
The current time is what Date() returns in the browser. 

When I type in 'new Date()' in my browser console it will return the correct time compensated for my time zone. Like this: "Thu Jan 30 2020 14:27:09 GMT+0100 (CET)"

AlbanT, 2020-01-30:
hmmm you are quite right...
When I do it in my browser (FireFox) I get the incorrect time: 
Date Thu Jan 30 2020 13:30:39 GMT+0000 (Coordinated Universal Time)
So it seems to be an issue in either my browser or OS :) 

I must admit that I have had similar issues in Node-Red

StefanCohen, 2020-01-30:
Ok. Good to know. I hope you will find a fix. I'm closing this issue.

closed 2020-01-30 by StefanCohen

AlbanT, 2020-01-30:
Problem was in Firefox.
privacy.resistFingerprinting was set to true