j7126/OctoPrint-Dashboard #124: [Req] Display Layer Notification Missing too often, please reduce frequency/ignore

ciordia9, 2020-02-05:
Really enjoy having a dashboard--I can't use the DisplayLayer plugin because it doesn't work in the devel branch and even when I did use the version you integrate I didn't really like it. However, your plugin constantly bugs me for it even when I have that part of the display unchecked.

"Can't get stats from DisplayLayerProgress. This plugin is required and provides GCode parsing for Fan Speed, Layer/Height info, Layer Durations and Average layer time. Is it installed, enabled and on the latest version?"

I'm all for checking to see if you can use things and integrate them but every time I turn around this error message is popping up. Can it be culled back a bit or ignored entirely?

StefanCohen, 2020-02-05:
Hmmm. I added the message when it became clear that too many users ignored the written requirements in the readme and filed tickets wondering why half of the dashboard doesn't work. I guess I could add a setting to disable the error message for users who know what they are doing.

OllisGit, 2020-02-17:
Hi @ciordia9 ,

what do you mean with "it doesn't work in the devel branch"? 
And what is the reason, why you don't like it?
You can turn off a lot of stuff in the plugin, see [DashBoard-Wiki](https://github.com/StefanCohen/OctoPrint-Dashboard/wiki/Help)

ciordia9, 2020-02-18:
Hiya @OllisGit, thanks for your efforts here.

[image]
[image]

Attaching a few screenshots. Running Rc4. LayerDisplayProgress does not compile right under this version I'm guessing since it hasn't been updated to handle the new requirements? As such I can't install it. That's fine, I don't 100% care about it. However, every time I reload, refresh, restart, the Octopi UI the popup reminding me that I need to have it installed is there. As you can see in the other screenshot, I do not have it active yet it still shows. (Unless user-error here and I've got some piece of it on that I shouldn't.)

ciordia9, 2020-02-18:
Circle back around because I hadn't looked much into this in a while. Also didn't realize you make the supplemental LayerProgress plugin as well. 

So-- current status was installed but disabled. I try to enable it, restart, its still disabled. I deleted, cleaned all, restart, reinstall, it looks like the install is normal. Restart. LayerProgress still disabled.

2020-02-18 12:33:45,761 > Successfully installed DisplayLayerProgress-1.18.0

I'm caught in a loop here. While again I don't need the details, I am trying to meet them but for some reason the plugin is just unhappy with something. Checked the logs and don't see anything out of the normal. If you have someplace you'd like me to check, I'd be happy to.

OllisGit, 2020-02-18:
Hi @ciordia9 ,
last weeks I tested a lot with OctoPrint 1.4.0rc3 Python2/3 without an installation issue.
Today I upgraded to OctoPrint 1.4.0rc5, DLP 1.18.0 and everything runs smoothly ;-)

Please tell me which python version do you use?

Also, switch the ```octoprint``` log level to ```debug```, reinstall the plugin and restart the server. 
Then look into the octoprint logfile. If you could not find any information why the plugin is not enabled, please create a new issue in my repository and attach the log file to it.
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues

I don't want to hijack this issue, because it is related to the popup-message and not the failed installation of DLP.

Thx,
Olli

ciordia9, 2020-02-18:
Thanks, opened and attached my log.

StefanCohen, 2020-03-04:
I have thought a bit more about this and decided to not implement a setting to disable the error message as I fear that it will lead to more unnecessary support.

closed 2020-03-04 by StefanCohen