j7126/OctoPrint-Dashboard #126: Hot end temp not displaying

Sarithz, 2020-02-12:
I'm running into an issue with the hot end gauge no longer working.  Reboots of the server do not help and I have had to uninstall and reinstall the plugin to get it working temporarily again.  It isn't an Octoprint issue as it's stats page does show the temperature correctly.  See the screenshots attached.
[image]
[image]

StefanCohen, 2020-02-12:
That seems strange. The temps are coming from the same viewmodel as the stats page. I will have to take a look to see if I can replicate the behaviour.

Sarithz, 2020-02-12:
Your plugin is awesome, and I've no issues at all until about a week ago.  Let me know if anything is needed from my end.

StefanCohen, 2020-03-03:
Hi again and sorry for the delay. I haven't been able to reproduce this on any of my systems and I haven't received any other reports on this issue. Are you still experiencing problems with this?

StefanCohen, 2020-03-06:
I'm closing this for now. Please re-open if you still have the issue.

closed 2020-03-06 by StefanCohen