j7126/OctoPrint-Dashboard #127: DLP deprecation of "totalHeightWithExtrusion"

OllisGit, 2020-02-17:
Hi,

I saw in your implementation that you use "totalHeightWithExtrusion", because totalHeight was not available.
https://github.com/StefanCohen/OctoPrint-Dashboard/blob/master/octoprint_dashboard/static/js/dashboard.js#L26

Since Version 1.17.0 the "totalHeight"-attribute is now the correct way to read the height. The calculation is related to the "height mode" in Plugin-Settings.

Please change your plugin to the correct attribute, so that I can remove the other attribute.

Thx, in advance
Olli

StefanCohen, 2020-03-03:
I finally have had time to set up a new dev env so this will be fixed in the next release.

StefanCohen, 2020-03-04:
Fixed in 1.11.2

closed 2020-03-04 by StefanCohen