j7126/OctoPrint-Dashboard #128: Top Margin adjustment.

Pheelix, 2020-03-04:
So for those of us that use Floating Nav Bar.  I had a issue when in fullscreen, The Navbar would cover some of the top icons for cpu, memory, disk and the 2 buttons for fullscreen and Full Window Browser.  I fixed this by adding margin-top: 30px; to 3 parts of the dashboard.css

.dasboardGridContainer {
  display: grid;
  grid-template-columns: auto auto auto;
  padding: 0px;
  margin-top: 30px;  << added
}

#fullScreenIcon {
  width: auto;
  height: 15px;
  float: right;
  -webkit-user-drag: none;
  margin-top: 30px; << added
}

#fullWindowIcon {
  width: auto;
  height: 15px;
  float: right;
  -webkit-user-drag: none;
  margin-top: 30px; << added
}

This will also effect the main screen when not in fullscreen.  But it is pretty minor and shouldn't effect any visuals.

If you could add this in a future update or maybe a fix it so the whole thing below the floating nav.

Also it would be cool if you could make it so stuff is drag able so we can move things around and customize it to the users liking.

Thanks.

StefanCohen, 2020-03-04:
Hmmm. I have never used the floating nav bar before but I can reproduce it now. My concern is that the floating nav bar is displayed at all when in full screen or full browser mode. The purpose of those two modes is to only present the dasboard content and strip away everything else (including the nav bar). Do you see my point? I'm currently thinking that I  should try to hide the floating nav bar when the dasboard is in full screen or full browser mode instead. Do you agree?

Pheelix, 2020-03-04:
You could make it a option to hide the floating navbar in the dashboard settings.  I still like the floating navbar in fullscreen because it allows me to click settings and other things listed on it that are not on the dashboard.  I would of just used themeifly to fix the margin issue, but I couldn't find the right div selector to do it.  But in the end it is your plugin so do what you think is best.

StefanCohen, 2020-03-04:
I see what you mean but I fear that the next request will be add a setting to show something else in fullscreen (like the sidebar or the tab bar ;-) ) and then we are kind of back to square one.

StefanCohen, 2020-03-04:
Update: There is a setting in the floatingnavbar plugin called Height Buffer. It appears to reserve some space under the navbar and seems to work just fine with the Dashboard. Please try it out to see if that works for you.

closed 2020-03-04 by StefanCohen

Pheelix, 2020-03-04:
ya I have tried adjusting that and made no difference.  It's ok tho, editing the css works fine no need to add or change anything if you don't want to.  It was just a suggestion really for those that do use floating nav to change the margin a little.  Keep up the good work :)

Also did you look into maybe adding the drag and drop type elements to move stuff around the dashboard?

StefanCohen, 2020-03-04:
Ok. Height buffer works fine for me in three different browsers so I think that may be the best option. 

Re drag&drop: https://community.octoprint.org/t/dashboard-v2-ideas/12717/19 Still just prototyping at this point. 

That bug may be a browser cache issue. I will have to try to replicate it.

Pheelix, 2020-03-05:
For the color bug I can send some screenshots or a video if it helps.  As far as the margin issue it could just be on my end I will do some testing with other browsers if it helps.  If you need a tester for the drag & drop let me know. :)

StefanCohen, 2020-03-06:
I have thought about this a bit more and I don't think there is anything I can do re the color bug as all I provide is the default css just like any other plugin with a tab view. This may be something related to Themeify but I'm honestly not sure about that either.

Pheelix, 2020-03-06:
Ya it could be a bug with themeifly.  Between themeifly, .css files and octoprint appearance.  That is 3 things trying to change colors of it.  But it does sound like a issue with themeifly as that is the plugin changing the colors.