j7126/OctoPrint-Dashboard #129: v1.11.2 repeatedly requires update

Taomyn, 2020-03-04:
**Describe the bug**
Had the alert to update to v1.11.2 but after restart of OctoPrint it says it's still needed - on second attempt I did see it "force" the install as it detected there was really no change.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on update alert
2. Restart OctoPrint

**Expected behavior**
No more prompts for update

**Screenshots**
[image]
[image]
[image]

**Files (please attach the following Files or screenshots when applicable):**
```
2020-03-04 09:03:10,498 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-03-04 09:03:58,040 - octoprint.plugins.softwareupdate - INFO - Starting update of dashboard to 1.11.2...
2020-03-04 09:04:12,907 - octoprint.plugins.softwareupdate - INFO - Update of dashboard to 1.11.2 successful!
2020-03-04 09:04:13,378 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-03-04 09:04:13,382 - octoprint.plugins.softwareupdate - INFO - Restarting...
2020-03-04 09:04:13,533 - octoprint.server - INFO - Shutting down...
2020-03-04 09:04:14,178 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-03-04 09:04:14,188 - octoprint.events - INFO - Event loop shut down
2020-03-04 09:04:14,196 - octoprint.server - INFO - Goodbye!
2020-03-04 09:04:16,259 - octoprint.startup - INFO - ******************************************************************************
2020-03-04 09:04:16,261 - octoprint.startup - INFO - Starting OctoPrint 1.4.0rc6
2020-03-04 09:04:16,262 - octoprint.startup - INFO - ******************************************************************************
2020-03-04 09:04:16,791 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online

```
**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Raspbian Buster
 - OctoPrint version: Version 1.4.0rc6
 - OctoPi version: 
 - Dashboard version 1.11.1
 - DisplayLayerProgress version 1.18.0

StefanCohen, 2020-03-04:
My bad. I didn't add the correct version identifier in 1.11.2 so Octoprint thinks that the newly installed version is still 1.11.1 and thus the loop. This will be fixed in 1.11.3

StefanCohen, 2020-03-04:
Just released 1.11.3 to the plugin repository.

closed 2020-03-04 by StefanCohen

Taomyn, 2020-03-04:
Confirmed, many thanks

[image]