j7126/OctoPrint-Dashboard #130: "Average Layer Time" shows only a value (in seconds?) after update

energiebrand, 2020-03-05:
Yesterday 5th of March 2020 updated the plugin to the latest version. After this during all prints instead of showing hours/minutes/seconds it just shows a value (maybe seconds). 
Version is Octoprint 1.3.12 on OctoPi 0.17.0 (Raspberry Pi 3B+).
I did a reset (did not change any value before) but no change.

[image]
[image]

StefanCohen, 2020-03-05:
Thanks. I see what the problem is and will provide a fix.

energiebrand, 2020-03-05:
Started a print on a different printer with another Octoprint server RPI 3B on Octoprint 1.4.0 on Octpi 0.16.0 with a different issue. It shows no Avergae Layer Time but a strange value below ETA.

Here Estimated Total Print Time is shown.

[image]
[image]

energiebrand, 2020-03-05:
Ups...after some more minutes waiting it also shows now a value:
[image]

StefanCohen, 2020-03-06:
The avg layer time is calculated by DLP after the first five layers according to the default setting so it will take some time to be displayed. The setting is configurable in DLP.

energiebrand, 2020-03-06:
ok, understood. But hope you will still work on the problem with the values :-)  If this needes some time until a solution will come maybe it´s better to downgrade to the previous release?

StefanCohen, 2020-03-06:
No worries. I have a fix. It was A typo introduced in 1.11.2.

Fixed in 1.11.4

closed 2020-03-06 by StefanCohen

energiebrand, 2020-03-06:
Thanks a lot! Just checked with 1.4.0 on 0.17.0 is working and back to live!
[image]