j7126/OctoPrint-Dashboard #134: Option to switch the logarithmic axis on Layer Durations graph or set max value while printing

StarNamer, 2020-03-28:
**Is your feature request related to a problem? Please describe.**
I quite often have the initial layer set to a slow speed, e.g. 20mm/s, and then switch to much faster, e.g. 100mm/s or occasionally higher. This has the effect that the initial layer time has on occasions bee 30 minutes or more. Later layers process in 1 to 2 minutes, This means the graph max is set to a high value but the graph drops to near zero within a few layers and any detail is hard to make out

**Describe the solution you'd like**
The option to toggle the graph between linear and log or simply display one or both

**Describe alternatives you've considered**
Alternative may be to set max value for time axis, but this still make detail hard to see when layer time gets shorter.

**Additional context**
Simulated (exaggerated) example from Excel. Spike at about layer 13/14 looks little different, but hump beteen layers 39 to 43 clearly visias taking noticably longer than other layers around the same time.
[image]

FYI, this a real graph to show the current effect:
[image]

StarNamer, 2020-03-28:
OK, I've looked at the code and can see the graph is a simple call to Chartist.Line. Unfortunately this doesn't have a log graph option that I can see, so the plugin might have to generate the data itself and fix up the axis labelling; e.g. take logs to base 10 so 10 becomes 1, 100 becomes 2, but the tick mark labels are "10", "100", etc..

More investigation needed...

StefanCohen, 2020-03-30:
Thanks for the feedback. This has been requested before but I haven't found any way to do this in chartist.js. Feel free to contribute a PR.

closed 2020-04-02 by StefanCohen