j7126/OctoPrint-Dashboard #136: [Feature Request] Python 3 Compatibility

jneilliii, 2020-03-31:
**Is your feature request related to a problem? Please describe.**
Running the latest version of OctoPrint 1.4.0 in a Python 3 environment plugin is not found in Plugin Manager, because of compatibility flags. I see the plugin is compatible in `__init__.py` but plugin repo has not been updated to list it as compatible.

**Describe the solution you'd like**
Update Plugin Repository with proper compatibility information.

**Additional context**
Please see [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready) for the required updates.

StefanCohen, 2020-03-31:
Thanks for the reminder. Pull request sent to plugin repository.

StefanCohen, 2020-03-31:
Merged

closed 2020-03-31 by StefanCohen