j7126/OctoPrint-Dashboard #138: Job status negative?

Protoncek, 2020-04-04:
Hello!
My job status is  ... negative...? (see picture). I'm printing face masks from Prusa. 
[https://www.prusaprinters.org/prints/27267-prusa-face-shield-stack-ver/files](url)
No matter, if i load pre-compiled gcode or i compile it myself. When i start to print job status is negative and "print time left" is zero. I have Prusa MK3S printer and in octoprint "M73override" plugin is installed - this plugin shows more accurate "time left", since prusaSlicer calculates this time.

Any clues?
[image]

StefanCohen, 2020-04-08:
Hi!

The Dashboard simply displays what is reported by OctoPrint. The error seems to be somewhere in the backend and out of my control as the job progress in the sidebar displays the same error. I suggest that you file an issue at the OctoPrint issue tracker.

closed 2020-04-08 by StefanCohen

Protoncek, 2020-04-08:
Thanks. It seems that the problem is in display layer plugin and the way prusaslicer does layer marking...