j7126/OctoPrint-Dashboard #140: Would love to be able to add the scond webcam to the from dashboard

mkenngott, 2020-04-10:
i love this plug in and i have always two web cams on my OctoPrint devcices and it would be helpfull if i can enable and see both.

Sembazuru, 2020-04-25:
I agree. I have 3 cameras on my R-Pi and manage them on the main Octopi interface with the MultiCam plugin. Maybe provide a way to use the MultiCam plugin to provide the camera switching logic (and thus multiple camera configurations only in one place)?

StefanCohen, 2020-04-25:
I have both tried and given up on adding more webcam streams but I would happily review a pull request if you would like to provide one.

Sembazuru, 2020-04-27:
I didn't see the first issue once I found this open. Reading the comments on Issue #65 I see you have given it the 'ole college try and have skin in the game of wanting it to work.

closed 2020-05-13 by StefanCohen

StefanCohen, 2020-06-02:
This has been implemented in v1.13.0 thanks to a patch provided by the user CynanX.