j7126/OctoPrint-Dashboard #143: Dashboard interferes with Fullscreen Plugin

AbuMaia, 2020-04-21:
**Describe the bug**
With the Dahboard plugin installed and active, the temperature and print time values do not display as they should when the webcam is set to fullscreen via the Fullscreen Plugin by @BillyBlaze.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Fullscreen Plugin (0.0.4)
2. Double-click webcam to enlarge, see the current temperature and print time at the bottom of the screen.
3. Install Dashboard (1.11.4)
4. Again double-click webcam to enlarge, see the temperature text is completely missing, and the print time section never shows any digits.
5. Disable Dashboard
6. Finally double-click webcam to enlarge, see the temperature and print time are back.

**Expected behavior**
Plugins should not cause other plugins to break.

**OctoPrint server and Plugins:**
OctoPrint 1.4.0 running on OctoPi 0.15.1 

**Desktop (please complete the following information):**
Ubuntu 18.04.4 LTS
Firefox 75

StefanCohen, 2020-04-25:
I can reproduce it but I can't figure out how the Dashboard is able to interfere since the values are read straight from the viewmodel. The Fullscreen plugin seems unmaintained as it hasn't been updated in two years and I haven't been able to reach the author.

BillyBlaze, 2020-04-27:
I have investigated this issue. The issue is caused by dashboard creating a button with the id 'job_pause'. This is the same id that is used by OctoPrint, so therefor the fullscreen plugin copies the wrong element.

I suggest using unique identifiers.

Dashboard code: https://github.com/StefanCohen/OctoPrint-Dashboard/blob/master/octoprint_dashboard/templates/dashboard_tab.jinja2#L49
OctoPrint code: https://github.com/OctoPrint/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/templates/sidebar/state.jinja2#L30

p.s. the plugin is still maintained, but since its very simple; it doesn't need any updates.

StefanCohen, 2020-04-28:
Thanks for the update. I will take a look at that.

AbuMaia, 2020-04-30:
You can close this issue. The latest release from Fullscreen has resolved the problem.

closed 2020-05-15 by StefanCohen