j7126/OctoPrint-Dashboard #145: Persistant statistics

Sembazuru, 2020-04-25:
**Is your feature request related to a problem? Please describe.**
When the print finishes all the print-time related statistics vanish. Can never see the print time of the last layer on the layer time graph, and if the print finishes when the printer is unattended (like in the middle of the night) all the statistics are lost.

**Describe the solution you'd like**
I'd like all the just completed print statistics to stay up on the dashboard until either:
* A new print is started with the currently loaded print file.
* A new print file is loaded.
* The server is reset.

It might also be nice to have the dashboard populated with the print statistics of the last time a specific file was printed when that file is loaded into Octoprint. And a way to download the print statistics to a computer if one wants to separately compare print to print.

StefanCohen, 2020-04-25:
Hmm. The purpose of the Dashboard was always to provide an insight into the current state of the printer. "A dashboard tab for Octoprint that displays the most relevant info regarding the state of the printer and any on-going print jobs." The only real deviation from that is the layer times graph that provides a job history during the print. I'm not sure what else besides the layer times would be interesting after the print is finished? Maybe this is better suited for some kind of print history plugin?

Sembazuru, 2020-04-28:
Specifically the most useful (for me, at least) pieces of information to be persistent after a print finishes are:

- Layer times graph
- File name _(I'm surprised this isn't shown as soon as a file is loaded, especially since the start button is on the dashboard. If the dashboard is in fullscreen, how do you know if a gcode is loaded before pressing the start button?)_
- Extruded filament
- Current print time _(once the print is finished, this would actually indicate the actual print time)_
- Average layer time

I was originally thinking my request would be to have all the statistics persist at the end so you don't have to have two types of settings, persistent and "clear at end of print". Unless there is something that I don't know about in octoprint that flushes your variable constructs, it seems to me it is up to you as to when you re-initialize the variables/arrays. Though, I haven't looked through your code, and I'm not sure if I'd [grok](http://catb.org/jargon/html/G/grok.html) your code if I did.

I understand that remembering the stats for past prints (and handling the garbage collection when a gcode files are deleted) is probably exceptionally complicated, so that part was more of a pipe dream.

But, since your plugin is already collecting the data for the Layer times graph, and supposedly storing as an array or list, as long as the data is available it would be nice to be able to download it as a .csv file. Shouldn't need a separate plugin to collect the same data that your plugin is collecting.

Here is a screenshot of the Dashboard with the items in the above list highlighted:
[image]

StefanCohen, 2020-04-28:
Ok. There is a print history plugin being developed by Olli (the author of DisplayLayerProgress). It seems to have all the features you want including CSV download. I think that would be a better starting point as it is purpose built for this use case including all the plumbing for persistent storage, etc. That plugin would support all past print jobs and not just the latest one. He also has access to all the layer durations as DLP is my source for that info so it should be feasible to add a layer graph similar to the one in the Dashboard.

https://github.com/OllisGit/OctoPrint-PrintJobHistory

Sembazuru, 2020-06-06:
I fear that this discussion got derailed from my primary request to one of the bonus ideas. The saving of stats wasn't the primary request. My primary request was to not hide some/any of the widgets when the print finishes. I know the data persists because sometimes when starting a new print I see the cleared widgets appear before the data is cleared and I see (for example) the layer time graph from the last print briefly. Even if the data is only stored in my browser's cache, if the widgets aren't hidden at the end of a print they should continue to show what is in the cache (whether the cache being shown is in the client or the server, doesn't really matter).

Should I enter a new issue to reset to my original request and none of the extra features that will be provided with the PrintJobHistory plugin, or can you re-open this issue?

StefanCohen, 2020-06-06:
No need. I will re-open and re-evaluate it.

StefanCohen, 2020-06-06:
Ok. I have been thinking about this for some time now and I keep falling back to my previous statement:  The purpose of the Dashboard is and has always been to show the _current_ state of the printer and any on-going print job. Yes, Some info from the last job is only cleared on print start (or plugin re-load) so it is technically there and available but I still don't see how it would make sense to display it in the dashboard when the printer is idling two days after the last print. That sort of goes against the principle to provide an insight to what's happening "right now". I think it would confuse a lot of users If the Dashboard for example shows the printer as offline and the layer graph or ETA at the same time. Is it still printing or has it finished now? I still think the PrintJobHistory plugin is a much better alternative to show historic job data as all of the above data is there. So no. I will not implement this request. But I encourage you to fork the project if this is something you really want to have and share with others.

Regards,
Stefan

closed 2020-06-08 by StefanCohen