j7126/OctoPrint-Dashboard #146: Gauge colour change not working.

mselkin, 2020-04-26:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
I am finding that the gauge colour change option only works when the temperature is below the set point (then the gauge is blue and turns green when the set point is reached). However, the gauges never turn red when the temperature is above the set point (at any threshold I tried).

**To Reproduce**
Steps to reproduce the behavior:
Manually raise hotted temp to 210, then set hotted temp to 200 in octoprint
**Expected behavior**
Gauge should be in red until temp falls to 200 c
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
* Browser console log
* gcode file that can be used to reproduce the problem

Any help would be appreciated,

Mark

StefanCohen, 2020-04-27:
Thanks. I can reproduce this and must have messed up something. I will take a look and try to fix it.

StefanCohen, 2020-05-15:
Fixed in 1.12.0

closed 2020-05-15 by StefanCohen