j7126/OctoPrint-Dashboard #148: CPU Temp for RasPi shows as 0 degC

spiff72, 2020-05-03:
[octoprint (3).log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4569629/octoprint.3.log)

**Describe the bug**
Temperature in Dashboard for the CPU (next to the CPU usage) is showing as 0 deg C.

**Expected behavior**
I expect to the the correct temperature.  I have another plugin (Navbar Temps) that correctly shows it.  I turned this off in the Navbar temperature plugin to see if it conficted, but it didn't help.

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log ATTACHED (I had to shorten it)

**OctoPrint server and Plugins:**
 - Platform: RPi 4 (but was seeing this on a Pi 3B+ previously)
 - OS and version: Raspbian Buster
 - OctoPrint version: OctoPrint 1.4.0
 - OctoPi version: OctoPi 0.17.0
 - Dashboard version 1.11.4
 - DisplayLayerProgress version 1.20.1dev

**Desktop (please complete the following information):**
 - OS and version:  Windows 10
 - Browser and version Chrome 81.0.4044.129

**Additional context**
Add any other context about the problem here.

StefanCohen, 2020-05-12:
Hmm. I haven't been able to reproduce this on my RPi 3b+ or my RPi 4 boards. Both report the temp correctly. Are the RAM and Disk usage reported correctly for you?

spiff72, 2020-05-12:
Yes - RAM and Disk show up fine.  And as can be seen in the attached screen shot, the CPU temp shows up in the navbar without issue, so this is very odd.
[image]

StefanCohen, 2020-05-13:
All the RPi info is coming from the same library but there may be sensor differences between board revisions that I'm not aware of. What board revisions are you using?

spiff72, 2020-05-13:
This is the info from the cat /proc/cpuinfo command (with serial number removed):
`Hardware        : BCM2835
Revision        : b03112
Serial          : -
Model           : Raspberry Pi 4 Model B Rev 1.2
`
I believe that this is the RasPi4 model with the "fix" in place for the USB-C power issue that was discovered in the early boards.

I can run other commands on it if you need more info...

Also, I am using several GPIO pins on the board for various plugins - namely for enclosure plugin and relay controls.  Not sure if that would interfere at all.  I will check my old Octoprint server (pi3B) and plug it in to check it as well (and see if it is reporting the temp correctly).

spiff72, 2020-05-13:
Well - interestingly, I just plugged in the Pi3B and connected to the octoprint server interface (still installed on the SD card and uses a different IP than my new one).  This one DOES show the temp.  Previously it did not, so now I am wondering if this is related to the GPIO that was connected previously.  The connections were the same as they are on the new pi4.  

If this is useful, here are the GPIO pins that are in use on my Pi4 (BCM numbering, and not including any 5V, 3.3V, or GND connections):
14, 15 - TxD and RxD for Direct Serial connection to my printer (Prusa MK3S)
18 - Used as a PWM control for enclosure fan - enabled, but NOT currently used
23 - Part of the serial cable connection to the printer
22 - Part of the serial cable connection to the printer
16 - Relay Control pin for enclosure plugin (turns exhaust fan on/off)
17 - Used as the 1-wire bus pin for my temperature sensor for enclosure.  4.7k pullup to 3.3V

Note:  All connections above the are described as part of the serial connection are derived from the instructions that Prusa provides for connecting a Pi Zero W directly to the control board of the printer (TxD, RxD, GPIO 22,23).  I don't know what purpose the 22,23 pins serve.

These are links to the two instruction pages for this process (I just substituted a modified cable and a Pi4 for the Pi Zero W):
https://help.prusa3d.com/en/article/raspberry-pi-zero-w-preparation-and-installation_2180
https://help.prusa3d.com/en/article/octoprint-building-an-image-for-raspberry-pi-zero-w_2182

These instructions include adding the following to /boot/config.txt:
`dtoverlay=pi3-miniuart-bt`

and modifying /boot/cmdline.txt to remove the serial console by deleting the following:
`console=serial0,115200`

spiff72, 2020-05-13:
One more thing - I looked through your code, and see that you appear to be using psutil to read temps.  Just for grins, I tried running the command that appears to pull the cpu temp on my active Pi4 server and got the following:

{'w1_slave_temp': [shwtemp(label='', current=26.125, high=None, critical=None)]}

I immediately recognized this as the temperature output of my enclosure temperature sensor  - a "one-wire" Dallas 18B20.  Unfortunately, I don't know how to proceed from here.  This may be a conflict with the enclosure plugin and/or the one-wire temp bus.  I suspect that the old pi3 server works because the temp sensor isn't connected and wasn't detected.  (I also borked that server trying to access the Octoprint virtualenv (but that's OK - I don't use it anymore, anyway!)

spiff72, 2020-05-13:
Last thing for now:
It appears that the Navbar Temp Plugin uses a different method to read SoC temp for raspi:
self.temp_cmd = '/opt/vc/bin/vcgencmd measure_temp'

It appears that there is code in there that establishes different methods for checking temps based on the hardware it is running on.  This gets beyond what I am capable of attempting to fix, though...

StefanCohen, 2020-05-13:
Ok. This is interesting. I'll focus on the RPi4 for the time being. My board revision is 1.1 so you do have a newer revision (1.2)

What is the output of print(psutil.sensors_temperatures(fahrenheit=False))? Do you only get the GPIO sensor in the output?

spiff72, 2020-05-13:
I don't think this is RPi4-specific, since I am positive my Pi3 did the same thing with the enclosure temp plugin installed and the Dallas temp sensor connected.

Here is the output of the command (no cpu-thermal):
```
>>> import psutil
>>> print(psutil.sensors_temperatures(fahrenheit=False))
{'w1_slave_temp': [shwtemp(label='', current=26.0, high=None, critical=None)]}
>>>
```

StefanCohen, 2020-05-13:
Ok. I have just installed a DHT11 on my RPi 4 and is booting it up. Let's see how that works.

spiff72, 2020-05-13:
I am curious whether that sensor will have the same issue as it isn't a 1-wire bus sensor (to my recollection).  Just for reference, I get the expected output on my mothballed Pi3B:
```
>>> import psutil
>>> print(psutil.sensors_temperatures(fahrenheit=False))
{'cpu-thermal': [shwtemp(label='', current=56.92, high=None, critical=None)]}
>>>
```

Stone-Hedge, 2020-05-13:
Hi All.  I have a pi4 that also exhibits the behaviour of showing zerm degrees on the dashboard.

Is there any info I can provide to assist?

I also have an 18B20 running as an enclosure sensor.

StefanCohen, 2020-05-13:
The DS18B20 seems to be a onewire sensor too. Correct? The problem seems to be that psutil gets confused and returns the external sensor output instead of the cpu temp. I can't reproduce this with my DHT11 sensor so I'm kind of stuck at the moment.

Stone-Hedge, 2020-05-13:
> The DS18B20 seems to be a onewire sensor too. Correct?

Correct

> The problem seems to be that psutil gets confused and returns the external sensor output instead of the cpu temp.

If this was the case, wouldn't the dashboard plugin be showing the enclosure temp instead of zero degrees? Forgive me if I have misunderstood.

StefanCohen, 2020-05-13:
psutil.sensors_temperatures(fahrenheit=False) should return an array of temp sensors and I filter out anything that is not "cpu-thermal" (for RPi) or "coretemp" (intel). In spiff72s case, the "cpu-thermal" is not present in the response, only "w1_slave_temp". What happens if you install psutil and run the following in python?

>>> import psutil
>>> print(psutil.sensors_temperatures(fahrenheit=False))

Stone-Hedge, 2020-05-13:
`pi@octopi:~ $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> print(psutil.sensors_temperatures(fahrenheit=False))
{'w1_slave_temp': [shwtemp(label='', current=25.437, high=None, critical=None)]}
>>>
`

It displays only a DS18B20 reading.

spiff72, 2020-05-13:
I don't think the DHT11 is a technically a 1-wire sensor (at least in terms of the one-wire bus scheme that is used for the Dallas sensor).  The one wire interface needs to be enabled on raspberry pi's.  It even has a pre-assigned GPIO (which I had to change in my case because it conflicted with the fan hat I used).

See this for how the 1-wire is enabled:
https://github.com/vitormhenrique/OctoPrint-Enclosure
scroll down to the sections on the DHT and DS18B20 specifics.

spiff72, 2020-05-13:
not sure if it is helpful, but found this with some random google searching:
https://github.com/xbmc/xbmc/issues/14775

Stone-Hedge, 2020-05-13:
> not sure if it is helpful, but found this with some random google searching:
> [xbmc/xbmc#14775](https://github.com/xbmc/xbmc/issues/14775)

Interesting.  

Is there any reason why Dashboard can't pull Pi temp from vcgencmd measure_temp ?

Strikes me as a fool proof way of doing it, under the caveat that I am not being a fool and not spotting a reason why this is a bad idea!

spiff72, 2020-05-13:
I suspect it doesn't work for intel hardware (and maybe others), but I agree - it would probably be a good approach for Pi (the navbar temps plugin uses it, and adjusts for other hardware platforms).

StefanCohen, 2020-05-13:
I suspect that the problem lies within psutil or some underlying software. This may warrant an issue at their tracker. People run octoprint on everything that runs python so HW coverage is pretty hard. Supporting vcgencmd would require some kind of python binding and a ton of fallbacks for other platforms I guess.

spiff72, 2020-05-13:
Simple *(but probably not simple) question.  I am not very familiar with how github works, but is there a way for a user to copy the code to a separate git that I could tinker with myself (without messing with your original)?  Is that a fork?  Or a clone?  Or something else?

I am curious if I could make changes to the code and get something that worked "just for me" - and not worry about the support for other platforms?  I could use a good coding exercise to dig further into python...

StefanCohen, 2020-05-13:
Well. It is not that complicated. You fork the project to your own account. Then you git clone it to your dev environment (could be your RPi), play around with it until you are happy and then git push it back to your own repo. If you end up with something you might want to contribute back you can create a pull request back to my repo and I can review it and possibly merge it  but that is entirely optional.

spiff72, 2020-05-13:
OK - thanks.  I might give it a shot, but I am not experienced with coding.  For now, I am OK with it being zero - mainly because I can see the temp by looking at the navbar.

Thanks for looking at this!

StefanCohen, 2020-05-13:
No problem. I encourage you to give it a shot. It is a fun learning experience. The dev documentation is excellent and the community is great. The plugin tutorial is an good primer.

closed 2020-05-13 by StefanCohen

spiff72, 2020-05-17:
Not trying to beat a dead horse, but I wanted to ask a simple question.  I have been experimenting with the plugin on my end, and I strongly suspect that my inexperience with coding is causing issues here, but I was wondering if something like the following would be a possible solution to the issue here.

```
    def psUtilGetStats(self):
        temp_sum = 0
        thermal = psutil.sensors_temperatures(fahrenheit=False)
        if "cpu-thermal" in thermal: #RPi
            self.cpu_temp = int(round((thermal["cpu-thermal"][0][1])))
        elif 'soc_thermal' in thermal: #BananaPi
            self.cpu_temp=int(round(float(thermal['soc_thermal'][0][1])*1000))
        elif 'coretemp' in thermal: #Intel
            for temp in range(0,len(thermal["coretemp"]),1):
                temp_sum = temp_sum+thermal["coretemp"][temp][1]
            self.cpu_temp = int(round(temp_sum / len(thermal["coretemp"])))
        elif "w1_slave" in thermal: #Dallas temp sensor fix
            tempFile = open("/sys/class/thermal/thermal_zone0/temp")
            cpu_val = tempFile.read()
            tempFile.close()
            self.cpu_temp = int(round(float(cpu_val)/1000)
        self.cpu_percent = str(psutil.cpu_percent(interval=None, percpu=False))
        self.cpu_freq = str(int(round(psutil.cpu_freq(percpu=False).current, 0)))
        self.virtual_memory_percent = str(psutil.virtual_memory().percent)
        self.disk_usage = str(psutil.disk_usage("/").percent)
```
Basically I just added another elif to the psUtilGetStas function that checks the "thermal" variable to see if it contains "w1_slave", and if so, I attempted to grab the contents of the /sys/class/thermal/thermal_zone0/temp file and use it to set the self.cpu_temp variable.  I am still trying to figure out the whole plugin/python/PyCharm IDE integration, so I have not been able to figure out how to debug the above code, since I am running Pycharm on an ubuntu laptop, and wasn't sure how to get the octoprint install and plugin functionality running correctly on a pi (since I need the pi to be able to read the file above).

I have tried installing the plugin manually with the above changes, and it doesn't appear to load correctly, so I suspect that there is an underlying issue with my code.  I don't see any dashboard tab, but the plugin appears in the Plugin Manager.

spiff72, 2020-05-17:
Followup to my above comment

I may have been missing plugin in the plugin settings because I was looking for the name "octoprint-dashboard" instead of just "dashboard".  HOwever, I am sure that the dashboard tab was missing when I was doing this.

Also (and more importantly), I just noticed that I was missing the closing parenthesis on the line:
`self.cpu_temp = int(round(float(cpu_val)/1000)`

I have now fixed this and the dashboard plugin loaded successfully, and displays a temperature on the testing pi that I had been trying things out on.  This does NOT have a dallas temp sensor connected so it theoretically would never get to the section of the if statement that I added.

I am going to try repeating the process by setting up another bash script in PyCharm to install the plugin to my "real" pi (with the temp sensor connected) and confirm that it really worked.

spiff72, 2020-05-18:
Well - I tried!  I loaded this plugin on my Pi4 with the dallas temp sensor connected, and it still reads "0" with my changes.  So I assume that I am missing something here.  If I could figure out how to debug with a live octoprint image and step through the plugin code I might get a better idea of where my method is failing.

spiff72, 2020-05-18:
...aaand I figured out my issue.  I needed to be checking for the thermal to contain "w1_slave_temp" (not just w1_slave).

This change now appears to be working on my system with the following as the psUtillGetStats function:

```
    def psUtilGetStats(self):
        temp_sum = 0
        thermal = psutil.sensors_temperatures(fahrenheit=False)
        if "cpu-thermal" in thermal: #RPi
            self.cpu_temp = int(round((thermal["cpu-thermal"][0][1])))
        elif 'soc_thermal' in thermal: #BananaPi
            self.cpu_temp=int(round(float(thermal['soc_thermal'][0][1])*1000))
        elif 'coretemp' in thermal: #Intel
            for temp in range(0,len(thermal["coretemp"]),1):
                temp_sum = temp_sum+thermal["coretemp"][temp][1]
            self.cpu_temp = int(round(temp_sum / len(thermal["coretemp"])))
        elif 'w1_slave_temp' in thermal: #Dallas temp sensor fix
            tempFile = open("/sys/class/thermal/thermal_zone0/temp")
            cpu_val = tempFile.read()
            tempFile.close()
            self.cpu_temp = int(round(float(cpu_val)/1000))
        self.cpu_percent = str(psutil.cpu_percent(interval=None, percpu=False))
        self.cpu_freq = str(int(round(psutil.cpu_freq(percpu=False).current, 0)))
        self.virtual_memory_percent = str(psutil.virtual_memory().percent)
        self.disk_usage = str(psutil.disk_usage("/").percent)
```

StefanCohen, 2020-05-18:
Nice. I usually do my development directly on an RPi. It tends to make things easier for me. I simply ssh to the RPi and edit using rsub/sublime text.

spiff72, 2020-05-18:
Thanks.  I attempted to submit a pull request for the changes I made here, but this is my first-ever attempt at such a thing, so I likely didn't do some of the things that I should have done.

spiff72, 2020-05-18:
In fact, now that I look at the "Software Update" section under the Octoprint settings, things look strange.  It doesn't ask me if I want to check for updates.  Maybe it knows that the Dashboard plugin is doesn't match your github repo - possibly since the version number is the same?  But I wouldn't have expected the other plugins to misbehave too...

spiff72, 2020-05-18:
I take back my previous statement - I just rebooted the pi - I think it may have just been offline for some reason.  The update buttons do show up now.

Stone-Hedge, 2020-05-18:
> Thanks. I attempted to submit a pull request for the changes I made here, but this is my first-ever attempt at such a thing, so I likely didn't do some of the things that I should have done.

Pull request looks fine to me, good work!