j7126/OctoPrint-Dashboard #150: Dashboard not displaying any values

Klammerson, 2020-05-11:
I installed the plugin properly without any issues. I also installed the not required but recommended plugin DisplayLayerProgress just to make sure.

The plugin isn't working for me at all. I can see the dashboard and can also access the settings but that's it. Changes in the settings don't appear in the dashboard itself. No values from the printer are displayed.

Uninstall and reinstall didn't help. Nothing in the log file.

I'm pretty clueless.

Thanks for your help.

StefanCohen, 2020-05-11:
Thanks for the feedback. Can you please post a screenshot of how the Dashboard appears during a print?

Klammerson, 2020-05-11:
Sure, will do. I have uninstalled it but once my print finishes I'll reinstall without DisplayLayerProgress and upload a screenshot.

StefanCohen, 2020-05-11:
DLP is required for most of the functionality in the Dashboard so please install it too. From the readme: "Dependencies: This plugin depends on DisplayLayerProgress to be installed. DisplayLayerProgress provides GCode analysis for the Fan, Layer, Height and Layer Average stats."

Klammerson, 2020-05-11:
Ok I will do so as well :) I will first install DisplayLayerProgress and restart. Then I will install the Dashboard and restart. Afterwards I will either bow my head in shame or send you a screenshot ;)

Klammerson, 2020-05-11:
What is also interesting is that the interface language changed from german to english. Not too bad but awkward. Still nothing to see.

[image]

StefanCohen, 2020-05-11:
Ok. That looks like a javascript error. Can you please bring up your browser console and copy any JS errors here?

Klammerson, 2020-05-11:
No errors there :(

[image]

Klammerson, 2020-05-11:
Using Chrome this came out!

Starting dependency resolution...
packed_plugins.js?62f87314:2124 [object Object]
packed_core.js?a314cc7f:14850 Could not instantiate the following view models due to unresolvable dependencies:
(anonymous) @ packed_core.js?a314cc7f:14850
packed_core.js?a314cc7f:14852 dashboardViewModel (missing: gcodeViewModel, enclosureViewModel )
(anonymous) @ packed_core.js?a314cc7f:14852
packed_core.js?a314cc7f:14861 ... dependency resolution done
packed_core.js?a314cc7f:15171 Initial application setup done, connecting to server...
packed_core.js?a314cc7f:12797 Connected to the server
packed_core.js?a314cc7f:15136 Finalizing application startup
packed_core.js?a314cc7f:15011 Going to bind 38 view models...
packed_core.js?a314cc7f:15064 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?a314cc7f:15064 Did not bind view model TelegramViewModel to target #wizard_plugin_telegram since it does not exist
packed_core.js?a314cc7f:15064 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?a314cc7f:15064 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?a314cc7f:3334 User klammerson logged in
packed_core.js?a314cc7f:15103 ... binding done
packed_core.js?a314cc7f:15115 Application startup complete
packed_plugins.js?62f87314:524 Array(1)0: "OctoPrint-"length: 1__proto__: Array(0)concat: ƒ concat()constructor: ƒ Array()copyWithin: ƒ copyWithin()entries: ƒ entries()every: ƒ every()fill: ƒ fill()filter: ƒ filter()find: ƒ find()findIndex: ƒ findIndex()flat: ƒ flat()flatMap: ƒ flatMap()forEach: ƒ forEach()includes: ƒ includes()indexOf: ƒ indexOf()join: ƒ join()keys: ƒ keys()lastIndexOf: ƒ lastIndexOf()length: 0map: ƒ map()pop: ƒ pop()push: ƒ push()reduce: ƒ reduce()reduceRight: ƒ reduceRight()reverse: ƒ reverse()shift: ƒ shift()slice: ƒ slice()some: ƒ some()sort: ƒ sort()splice: ƒ splice()toLocaleString: ƒ toLocaleString()toString: ƒ toString()unshift: ƒ unshift()values: ƒ values()Symbol(Symbol.iterator): ƒ values()Symbol(Symbol.unscopables): {copyWithin: true, entries: true, fill: true, find: true, findIndex: true, …}__proto__: Object Object
packed_plugins.js?62f87314:533 Object
packed_plugins.js?62f87314:524 Array(1) Object
packed_plugins.js?62f87314:533 Object
packed_plugins.js?62f87314:524 Array(1) Object
packed_plugins.js?62f87314:533 Object
packed_plugins.js?62f87314:524 Array(1) Object
packed_plugins.js?62f87314:533 Object
packed_plugins.js?62f87314:524 Array(1) Object
packed_plugins.js?62f87314:533 Object
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.preload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.postload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME

StefanCohen, 2020-05-11:
There are two unavailable viewmodels: gcodeViewModel, enclosureViewModel. You have probably disabled the gcode viewer in the Octoprint Settings. That setting must be enabled for the layer graph to work. You have probably also enabled the temp/humidity sensor in the Dashboard settings but the Enclosure Plugin can't be found. Try to disable that setting or install the Enclosure plugin. Those two things should fix your issue. Let me know how it goes.

Klammerson, 2020-05-12:
Thanks for the hint. After enabling code viewer and a fresh restart it seems to work. I didn't have to install the enclosure plugin since I don't use this feature.
Thanks for the quick and helpful support!

StefanCohen, 2020-05-12:
Perfect.

closed 2020-05-12 by StefanCohen