j7126/OctoPrint-Dashboard #151: DisplayLayerProgress toggle

CynanX, 2020-05-14:
Hi,

First, love you plugin - thank you for creating it!

Now.. you may have heard that the DisplayLayerPlugin is _apparently_ causing layer shift issues so many have currently disabled it. Whilst you can still use the Dashboard plugin it does pop up a warning each time I go into OctoPrint.

Could you add a control or toggle under it's settings to disable the parts that use this plugin for those that want to run it disabled?

It might be nice to also put in toggles for the other plugin dependencies too.

Thank you!

--Cynan.

daveindustries, 2020-05-14:
Hey Also came here to share the same exact same words as the op about DisplayLayerPlugin.

No need to re write other than to also say Dashboard is an awesome plugin thank you for making it!

D

StefanCohen, 2020-05-14:
Thanks for the feedback.

DLP provides the data for the Fan speed widget, Current/Total layers widget, Current/Total height widget, Average layer duration widget and the Layer duration graph. This is a substantial part of the Dashboard widgets. I can probably add a setting to disable the warning message (at least the popup) but disabling the widgets would be non-trivial as DLP has been a firm required dependency from the start (hence the warning message).

daveindustries, 2020-05-14:
I see, in that case just being able to disable the popup would be an awesome temporary solution.

It's been a few months since disabling DLP so hopefully it will be back working soon.

StefanCohen, 2020-05-14:
Ok.

CynanX, 2020-05-14:
Yeah. Not to bothered about the widgets being hidden, it's the warning annoying! Hopefully the DLP manages to solve the issue soon. Thank you.

StefanCohen, 2020-05-14:
Yeah. I highly doubt that the observed layer shifting is caused by DLP but that's a different story.

StefanCohen, 2020-05-15:
Fixed in 1.12.0

closed 2020-05-15 by StefanCohen