j7126/OctoPrint-Dashboard #153: Might want to check version numbers for the 12.0 build...

SwervingLemon, 2020-05-16:
Looks like the versioning is broken. Mine complains that version 12 is available, downloads, installs, tells me that version 11 something was successfully installed and then I get the notification that new version is available.
Got so annoying I uninstalled it completely and reinstalled. Says it installed version 11. This is on two different Octoprint instances on two different Pi3B+.

Taomyn, 2020-05-16:
Yup

`Updating, please wait.
+++++++++++++++++++++++++++++++++++++++
Now updating Dashboard Plugin to 1.12.0
+++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/StefanCohen/OctoPrint-Dashboard/archive/1.12.0.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/StefanCohen/OctoPrint-Dashboard/archive/1.12.0.zip
Downloading https://github.com/StefanCohen/OctoPrint-Dashboard/archive/1.12.0.zip
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-Dashboard==1.11.4 from https://github.com/StefanCohen/OctoPrint-Dashboard/archive/1.12.0.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Dashboard==1.11.4) (1.4.0)
Requirement already satisfied: psutil in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Dashboard==1.11.4) (5.6.7)
Requirement already satisfied: flask<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.12.5)
Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.56.0)
Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.4.1)
Requirement already satisfied: futures<3.4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (3.3.0)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (2.8.1)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.7.19)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.1)
Requirement already satisfied: chainmap<1.1,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.0.3)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.12.2)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (3.7.4.1)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.9.0)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.6.5)
Requirement already satisfied: emoji<0.6,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.5.4)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.2)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (3.1.1)
Requirement already satisfied: sentry-sdk==0.13.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.13.2)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (4.0)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (5.3.1)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (5.2.1)
Requirement already satisfied: werkzeug<0.17,>=0.16 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (0.16.1)
Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (7.1.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (3.4)
Requirement already satisfied: scandir<1.11,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.10.0)
Requirement already satisfied: monotonic<1.6,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.5)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (2020.2.20)
Requirement already satisfied: filetype<2,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.0.6)
Requirement already satisfied: blinker<1.5,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (1.4)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.11.4) (4.5.3)
[... cut ...]
Created wheel for OctoPrint-Dashboard: filename=OctoPrint_Dashboard-1.11.4-py2-none-any.whl size=166909 sha256=3f0f86212060b204b7613b9dfd54f6734bb881b40afe7a1f948ae8c331b37a17
Stored in directory: /tmp/pip-ephem-wheel-cache-S1Ji6V/wheels/7b/9b/d1/5f1ab3258dccc148ee8531014e58003157f6785e9c3613ef90
Successfully built OctoPrint-Dashboard
Installing collected packages: OctoPrint-Dashboard
Successfully installed OctoPrint-Dashboard-1.11.4
The update finished successfully and the server will now be restarted.`

somehibs, 2020-05-16:
Yep, still being prompted to update after updating.

themiskt, 2020-05-16:
Mine too.. !

StefanCohen, 2020-05-16:
Yeah. Sorry. I forgot to commit the changes to setup.py. Too rusty for this apparently. Should be fixed in 1.12.1

closed 2020-05-16 by StefanCohen

Taomyn, 2020-05-16:
Confirmed now ok, thanks @StefanCohen

SwervingLemon, 2020-05-16:
Yes, thank you!

chimcen, 2020-05-16:
OctoPrint tells me 1.12.0 is the latest...

chimcen, 2020-05-16:
It works now - had to ignore the update cache witzin octoprint...

Thank you :)