j7126/OctoPrint-Dashboard #160: Layer Durations Graph not shown

OllisGit, 2020-05-21:
Hi @StefanCohen ,

I received a ticket that the layer-duration is not visible (https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/148) 

I could reproduce the issue and I debugged till the frontend:
[image]

All layer duration-values were sent to the javascript and the title "Layer Durations" is also shown, but not the graph.

I used the following gcode with Dashboard V1.12.2 (master-branch) and DLP 1.21.0
```
M117 INDICATOR-Layer1
G4 S5
M117 INDICATOR-Layer2
G4 S10
M117 INDICATOR-Layer3
G4 S15
M117 INDICATOR-Layer4
G4 S10
M117 INDICATOR-Layer5
G4 S5
M117 INDICATOR-Layer6
G4 S3
M117 INDICATOR-Layer7
M117 finished
```

Any idea?

StefanCohen, 2020-05-25:
Strange. It seems to work just fine here. I'm running your gcode on Dasboard v1.12.2 and DLP 1.21.0 on OctoPrint 1.4.0 (OctoPi 1.17.0). This is my result in Safari:

[image]

Can this be a browser issue? What browser and version did you try?

StefanCohen, 2020-05-25:
Well. I have Tried the latest versions of Chrome and FF and it works there too so I'm stumped. The graph loads and is populated with the correct values regardless of what I try.

OllisGit, 2020-05-25:
hmm...okay I not a javascript-expert, but I will try to debug why the graph is not shown (maybe caching issue)

StefanCohen, 2020-05-25:
Does it throw any js errors when the graph doesn't load?

OllisGit, 2020-05-25:
Okay, found the issue!
The graph is not painted, because "layerLabels" is not transferred to the frontend:
```
if (data.layerTimes && data.layerLabels) { self.renderChart(data.layerTimes, data.layerLabels); }
```
and the reason why the data is not sent, is that ```self.psUtilGetStats()``` crashed in the ```send_notifications``` method.
```
  File "/Users/o0632/0_Projekte/3DDruck/OctoPrint/3rdParty/OctoPrint-Dashboard/octoprint_dashboard/__init__.py", line 57, in send_notifications
    self.psUtilGetStats()
  File "/Users/o0632/0_Projekte/3DDruck/OctoPrint/3rdParty/OctoPrint-Dashboard/octoprint_dashboard/__init__.py", line 31, in psUtilGetStats
    thermal = psutil.sensors_temperatures(fahrenheit=False)
AttributeError: module 'psutil' has no attribute 'sensors_temperatures'
```
I surrounded it with a try/expect and now I can see the same graph as in your screenshot.

In my environment it is clear, because I didn't use a Raspberry for testing.
I need to ask the user what kind of setup he used. 

But you should think about the try/expect, because not every "op-server" has build in sensors.

StefanCohen, 2020-05-25:
Thanks. I'll try to add some error handling around the call to psutil.

energiebrand, 2020-05-27:
This is my comment from the issue I opened in Display Layer Progress Plugin:

Octoprint 1.4.0 on Octopi 0.17.0 and Octoprint 1.4.0 on Octopi 0.16.0
I can confirm that issue #148 (Display Layer Progress) can be closed from my point of view.
**Also from my point of view issue #163 #164 of the Dasboard can be closed. I propose to close issue #160 in Dashboard.**

Octoprint 1.4.0 on Octopi 0.17.0
What did I?
Removed Dashboard, Display Layer Progress and Custom Control Editor including data.
Sudo apt-get update & upgrade (just routine...)
Installed Dashboard, Display Layer Progress and Custom Control Editor (all latest revisions)
Defined Custom control buttons in Custom Control Editor, saved, shut off, boot
Now everything works, all problems are gone!

Octoprint 1.4.0 on Octopi 0.16.0
Did the same as above but -> the issue #31 & 32 of the Custom Control Editor are still present
I did a fallback to the old release of Custom Control Editor; here it works!
However no more problems in combination with Dashboard and Display Layer Progress!

StefanCohen, 2020-05-27:
Allright. Closing this issue.

closed 2020-05-27 by StefanCohen