j7126/OctoPrint-Dashboard #162: show SNMP values in Dashboard

AndreasBrauckmann, 2020-05-23:
there is the possibility to display SNMP values in Dashboard, i have a power supply Microchip TCP / IP Voltmeter & Ammeter for two lamps, Raspberry and a USB hub. 

pi@octopi:~ $ snmpwalk -v2c -c public 192.168.1.2 iso.3.6.1.4.1.17095.3
iso.3.6.1.4.1.17095.3.4.0 = STRING: "5.59" -> (Volt USB Hub)
iso.3.6.1.4.1.17095.3.5.0 = STRING: "5.06" -> (Volt Raspberry)
iso.3.6.1.4.1.17095.3.6.0 = STRING: "9.90" -> (Volt Spots)
iso.3.6.1.4.1.17095.3.7.0 = STRING: "11.03" -> (Volt Light Bar)
iso.3.6.1.4.1.17095.3.8.0 = STRING: "23.5" (Temperature in the top of the case)

[image]

[image]

http://www.digital-measure.com/html/tcp_voltmeter_v13.html

Ikea lack table for prusa:

[image]

StefanCohen, 2020-05-25:
Thanks for the feedback. This sounds interesting but I'm not yet sure how to implement this in a generic way. I guess I will need to think about it for a while. How do you imagine the configuration would work given that there may be multiple sensors displaying different types of info? How would they appear on the dashboard in your opinion?

AndreasBrauckmann, 2020-05-25:
The easiest way would probably be via a file sensor or shell in the enclosure plugin. Simply display the values via a query. I think this is the easiest way to read in all possible values and you are flexible.

[image]

StefanCohen, 2020-05-25:
Well. I rely on the enclosure plugin for all other sensors (DHT11 for example) and it already supports shell commands. I'm thinking that it would already work to use Enclosure to invoke snmpget and then simply display that sensor in the dashboard. I think that would work without any modifications at all. At least for temperature sensors. I will play with that to begin with and see what I can learn from that.

StefanCohen, 2020-05-25:
Hmm. Never mind. Enclosure only supports shell comands for outputs. I will have to think about this for a while.

StefanCohen, 2020-06-04:
And one more thing. I will  have to add permissions to the config as running arbitrary shell commands like this should require admin access.

Willmac16, 2020-09-30:
With the addition of command widgets you should now be able to do this.

closed 2020-09-30 by Willmac16

AndreasBrauckmann, 2020-11-04:
Sorry for delay ... great Work Stefan, thanks a lot of.