j7126/OctoPrint-Dashboard #163: RPi data none more updated during printing

energiebrand, 2020-05-25:
I am running Octoprint 1.4.0 on OctoPi 0.16.0 on a Raspberry Pi 3B.

I noticed that RPi data are shown **before** print starts, but get frozen **after** printing starts (see picture 1).

When opening a new browser window during printing all RPi values are "0" (picture 2).

When opening the Resource Monitor it shows that data are delieverd continously (picture 3)

[image]
[image]
[image]

StefanCohen, 2020-05-25:
Thanks for the feedback. I haven't been able to reproduce your problem on my RPi 3b+ or my RPi4. Both shows the stats updating before, during and after a print job. The only things I can think of is if something else blocks psutil but I don't know how to check that other than running a debug version of the Dashboard on your RPi
.

StefanCohen, 2020-05-25:
Do you have any temperature sensors hooked up to your RPi via GPIO or other means?

energiebrand, 2020-05-26:
No, there is nothing connected to the GPIO´s. This behavoir was never seen before the last PlugIn Update. I always have seen a changing CPU clock and CPU % value during printing before.

energiebrand, 2020-05-27:
This is my comment from the issue I opened in Display Layer Progress Plugin:

Octoprint 1.4.0 on Octopi 0.17.0 and Octoprint 1.4.0 on Octopi 0.16.0
I can confirm that issue #148 (Display Layer Progress) can be closed from my point of view.
Also from my point of view issue #163 #164 of the Dasboard can be closed. I propose to close issue #160 in Dashboard.

Octoprint 1.4.0 on Octopi 0.17.0
What did I?
Removed Dashboard, Display Layer Progress and Custom Control Editor including data.
Sudo apt-get update & upgrade (just routine...)
Installed Dashboard, Display Layer Progress and Custom Control Editor (all latest revisions)
Defined Custom control buttons in Custom Control Editor, saved, shut off, boot
Now everything works, all problems are gone!

Octoprint 1.4.0 on Octopi 0.16.0
Did the same as above but -> the issue #31 & 32 of the Custom Control Editor are still present
I did a fallback to the old release of Custom Control Editor; here it works!
However no more problems in combination with Dashboard and Display Layer Progress!

closed 2020-05-27 by StefanCohen