j7126/OctoPrint-Dashboard #164: No "Extruded Filament" shown ("0" value shown during print)

energiebrand, 2020-05-25:
Same configuration as issue #163 but now I noticed that also the "Extruded Filament" stucks at "0".
[image]

StefanCohen, 2020-05-25:
Thanks for the feedback. Do you get any javascript errors when this happens?

StefanCohen, 2020-05-25:
I think Ollie found a possible explanation for this in #160 . His problem was that psutil crashed on his system causing the backend to stop sending data to the frontend. This is likely what causes #163 as well. I don't know why this happens on your RPi since I can't reproduce it here but I can try to add some error handling around the call to psutil to mitigate or at least detect and log the error.

energiebrand, 2020-05-26:
...no javascript errors... are shown.
Today checked on a RPI 3B+ with Octoprint 1.4.0 on OctoPi 0.17.0 with same behaviour as on RPi 3B  with Octoprint 1.4.0 on OctoPi 0.16.0

energiebrand, 2020-05-27:
This is my comment from the issue I opened in Display Layer Progress Plugin:

Octoprint 1.4.0 on Octopi 0.17.0 and Octoprint 1.4.0 on Octopi 0.16.0
I can confirm that issue #148 (Display Layer Progress) can be closed from my point of view.
Also from my point of view **issue #163 #164 of the Dasboard can be closed. I propose to close issue #160 in Dashboard.**

Octoprint 1.4.0 on Octopi 0.17.0
What did I?
Removed Dashboard, Display Layer Progress and Custom Control Editor including data.
Sudo apt-get update & upgrade (just routine...)
Installed Dashboard, Display Layer Progress and Custom Control Editor (all latest revisions)
Defined Custom control buttons in Custom Control Editor, saved, shut off, boot
Now everything works, all problems are gone!

Octoprint 1.4.0 on Octopi 0.16.0
Did the same as above but -> the issue #31 & 32 of the Custom Control Editor are still present
I did a fallback to the old release of Custom Control Editor; here it works!
However no more problems in combination with Dashboard and Display Layer Progress!

StefanCohen, 2020-05-27:
Ok. I'm happy that you solved the issue. Closing.

closed 2020-05-27 by StefanCohen