j7126/OctoPrint-Dashboard #166: "Floating Navbar" plugin does not work when Dashboard is in Full Browser Window mode

KenLucke, 2020-05-28:
Linking back to that ticket in the Floating Navbar plugin thread, as requested by the plugin author.  Details are there.

https://github.com/jneilliii/OctoPrint-FloatingNavbar/issues/7

jneilliii, 2020-05-28:
@StefanCohen I'm assuming we just need to tweak some js or css for this one, just can't really invest any time into it right now.

StefanCohen, 2020-05-28:
Ok. I will take a look and try to figure this one out.

KenLucke, 2020-05-28:
> @StefanCohen I'm assuming we just need to tweak some js or css for this one, just can't really invest any time into it right now.

I kinda ass/u/med that, and it's not a critical problem.  Just figured I'd bring it up for the back burner.

StefanCohen, 2020-05-28:
Yes. I've actually considered hiding the floating navbar entirely when in full browser or full screen mode as the purpose of those modes is to hide everything except the dashboard. But I realise that some users still want it there even in those modes and that's fine. I'm looking at it right now and am trying to figure out how my css reset works as that is the most likely culprit.

KenLucke, 2020-05-28:
> Yes. I've actually considered hiding the floating navbar entirely when in full browser or full screen mode as the purpose of those modes is to hide everything except the dashboard. But I realise [...]

Yes, I'd prefer to keep the navbar, as there are things in it that are not instantly accessible if it's not there, like my emergency shutdown button, the Settings link, etc.  Thanks for taking this up.

StefanCohen, 2020-05-29:
I looked at it yesterday and I couldn’t find anything obvious. I’m pretty bad at CSS so I’ll probably have to request some help with this.

jneilliii, 2020-05-30:
No problem @StefanCohen I'll see if I can figure it out this weekend. Is the assumption correct that when Dashboard goes into fullscreen mode that the navbar isn't supposed to be visible at all?

StefanCohen, 2020-05-30:
Thanks. No. It should be visible in both full screen and full browser mode (there are two separate buttons for each mode in Dashboard) and it is. The problem appears to be that the setting to reserve vertical space for the floating navbar is broken in full browser mode.

jneilliii, 2020-05-30:
I think this may be a Themeify conflict, the default settings don't appear to effect the default UI.

[image]

jneilliii, 2020-05-30:
So @StefanCohen I was wrong, I'm able to reproduce and I know how to fix it, but I need to figure out when you're applying the `dashboard-full` class to the dasboardContainer. Looking in your code now.

jneilliii, 2020-05-31:
lol...I just realized the container was name dasboadContainer instead of dashboardContainer. If you ever change that you'll need to let me know please.

jneilliii, 2020-05-31:
I've added a patch to my plugin that should resolve this @StefanCohen. Once @KenLucke confirms he can close the ticket here.

jneilliii, 2020-05-31:
@KenLucke has confirmed my update fixed this issue. You can close this one @StefanCohen.

StefanCohen, 2020-05-31:
Ok. Thanks!

closed 2020-05-31 by StefanCohen

StefanCohen, 2020-06-09:
@jneilliii I intend to fix the css typo in the upcoming Dashboard v1.14 release. It may take a week or so prepare the release and I'll try to remember to notify you when it is ready.