j7126/OctoPrint-Dashboard #169: Multiple instances of Dashboard

tedukes, 2020-05-29:
Hello,
Really enjoy using your dashboard plugin.
I have recently added another printer and have multiple instances of Octoprint running with multiple webcams.
It would be awesome if I could have Dashboard running on both instances.
Thank you!

StefanCohen, 2020-05-29:
Are you running two instances of Octoprint on the same host? That’s not something I’ve tried so I dont’t know what would happen. What problems are you running into?

tedukes, 2020-05-29:
Hi,

 

Right, I’m running 2 instances on a single RPi. I followed Chris Riley’s video on Youtube to get it working. https://www.youtube.com/watch?v=7Saa1HpLRoM

 

The only issue is its only running on the first instance. No dashboard on the second. I’m not sure why its not available as other plugins are.

 

Thanks!

StefanCohen, 2020-05-30:
Can you please attach the octoprint log for the second instance?

jneilliii, 2020-05-31:
I've had to deal with this issue before @StefanCohen. Any places where you are making API calls or loading URLs from your plugin side, you need to make the URL relatives to the global variables API_BASEURL or BASEURL.

tedukes, 2020-05-31:
Hello,

 

I don’t believe there is a log for the second instance, unless its somewhere I’m not aware of. I have attached /var/log/message and /var/log/webcam2

 

Thanks!

KenLucke, 2020-05-31:
> Are you running two instances of Octoprint on the same host? That’s not something I’ve tried so I dont’t know what would happen. What problems are you running into?

Used to work (v1.11.4)  Now, it only shows the webcam on one of the instances...  If I reload one, it starts showing the webcam on the other, and loses it on the one reloaded.

https://www.dropbox.com/s/lbfrwkv2womwchj/screencast%202020-05-30%2020-15-49.mp4

Started about the time you and the author of FloatingNavBar started tinkering due to my request over there.  Disabling/Removing FloatingNavBar does not change the behavior.

tedukes, 2020-06-01:
https://www.dropbox.com/s/lbfrwkv2womwchj/screencast%202020-05-30%2020-15-49.mp4

I don’t use that plugin.

Getting two webcams working was tough. Watched Chris Riley’s video numerous time. I pretty much gave up. Could see both cams on the first instance but nothing but a snapshot on the second. One day, it just started working by itself. Crazy! 

Using a Monoprice powered hub.

KenLucke, 2020-06-01:
> Getting two webcams working was tough. Watched Chris Riley’s video numerous time. I pretty much gave up. Could see both cams on the first instance but nothing but a snapshot on the second. One day, it just started working by itself. Crazy!

Not using two webcams.  Same webcam, two instances of the same OcctoPrint.

StefanCohen, 2020-06-01:
I have to say that I'm quite confused. I still don't understand what the problem is. @KenLucke Please create a new issue as your problem doesn't seem to be related to running multiple octoprint instances.

@tedukes I don't have much to go on here. Can you please try to describe the problem in detail? Is the Dasboard tab completely missing in the second instance? I will also need an octoprint log file for each of the instances to begin digging.

tedukes, 2020-06-02:
OK, I found the logs I think you’re asking for. Also I’m attaching 2 screen prints of the two instances. Hope the files aren’t too large.

Thanks!!

StefanCohen, 2020-06-02:
Thanks for the update. Please attach the logs on the Github page. It seems like they are lost when replying via email.

tedukes, 2020-06-02:
here ya go
[image]
[octoprint_logs.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4717923/octoprint_logs.zip)
[image]

jneilliii, 2020-06-02:
Still no logs @tedukes?

tedukes, 2020-06-02:
Posted on Github. look between the 2 pics.

Dashboard (#169)

StefanCohen, 2020-06-02:
It seems like both Dashboard and DLP are disabled in the second instance. Can you try to enable them in the plugin manager?

tedukes, 2020-06-02:
Thanks!! That was the problem. Didn't occur to me it needed to be enabled.

StefanCohen, 2020-06-02:
No problem. Closing this issue then.

closed 2020-06-02 by StefanCohen